tzapu/WiFiManager #905: Pharse this lib to esp-idf

Nathan-ma, 2019-06-27:
Guys it's not actually an issue but a request, I'm a big fan of this library! great work indeed.

Can you guys make this wonderful work to run on esp-idf. I mean 100% C++ based, no arduino framework used.

The reason why i'm asking this is, esp32 AP mode seems to be broke on the current version of arduino-esp32. 

Would be amazing to use such an awesome lib on esp-idf.

tablatronix, 2019-06-27:
ap mode is broken ? I have no reports or issues myself, have a reference?

Nathan-ma, 2019-06-27:
never mind, I was using an extarnal link on html and it was freezing the whole thing, got it fixed already using html libs on spiffs. anyway you should think on making wifi manager for esp-idf