tzapu/WiFiManager #908: Use WiFiManager only on first boot

array81, 2019-07-04:
On my project I need do it:

1. On first device start I need use WiFiManager to connect find and connect to WIFI;
2. On other device start I don't want use WiFiManager but get SSID and password saved by WiFiManager and try to connect with these data until connect is established;

My device have a reset button to delete all settings (also WIFI settings) so next start WiFiManager must be run.

Is there a way to know if WiFiManager just have stored SSID and password and get these data?
In this way I can enable and disable WiFiManager on start.

tablatronix, 2019-07-04:
you just call esp begin(), esp stores the credentials

array81, 2019-07-04:
Sorry I don't understand. Can you explain to me.

tablatronix, 2019-07-04:
wifimanger does not save anything, esp does, so just use esp functions, also autoconnect already only lanches the captiveportal if the connection fails.