tzapu/WiFiManager #911: AP appears despite  of entering client mode.

YogKar, 2019-07-11:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] travis

**Esp8266/Esp32:**
- [ ] ESP8266

**Hardware:			

- [ ] ESP12 E/F/S (nodemcu, wemos, feather)

### Description

Using https://github.com/tzapu/WiFiManager/tree/travis

I need non blocking AP, and so i am using "https://github.com/tzapu/WiFiManager/tree/travis"
I was testing the basic Autoconnect .
When I configure , save New AP in portal, and restart ESP8266 module, The device gets configured in client mode and connects to the configured AP. But along with that it also Shows AP which is typically ESPxxx (starting with ESP). which ideally should have not been appearing.
to check further I tried to connect it. but could not get the config portal page. and this happens in blocking as well in non blocking mode.

what must be happening? How should this be resolved?

https://github.com/tzapu/WiFiManager/issues/457
The above issue looks similar . Is it resolved ?

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2019-07-11:
WiFi.mode(WiFi_STA); // set mode explicitly
Esp does that by default

closed 2019-07-11 by tablatronix

YogKar, 2019-07-12:
It worked. 
Thanks for working on nonblocking and sharing it. It's a huge effort saver.

tablatronix, 2019-07-12:
I assume you meant you are using 
https://github.com/tzapu/WiFiManager/tree/development

not travis. ?

YogKar, 2019-07-13:
"https://github.com/tzapu/WiFiManager/tree/development" shows that the "build failing" for both ESP8266 and ESP32. So i thought it's somewhere incomplete or broken. 

But after your reply I tested it.  Its working flawless.

tablatronix, 2019-07-13:
the script needs to be fixed, it broke when something changed in esp toolchain, I have not had time to look at it

YogKar, 2019-07-13:
I have to add few things so that other visitors here get some help and if I have done wrong you can correct me. This is based on my experience.
1) polling process() every 10ms gave me best results while using non-blocking mode..
2) I stopped polling when the device is configured in station mode (client). 
Makes sense?

tablatronix, 2019-07-14:
Yes you will have to process the web server constantly or it will not handle requests fast enough, and probably turn debugging off and serial prints.

You can process when ever you want the library will return if the webserver is not running, you can also run the webserver in sta mode without the ap or without the captive portal

YogKar, 2019-07-16:
Sure.