tzapu/WiFiManager #912: Would it be possible to not let autoConnect block the setup?

markg85, 2019-07-13:
Hi,

Currently the autoConnect function blocks the setup till it either makes a wifi connection or sets itself up as AP in case it couldn't make a connection.

That on it's own is awesome!

However, i'm now playing with a usecase where i want to fade a light in as soon as the device turns on. That "fading in" is something that's run in the loop function. And that loop function only starts when the setup is done... which autoconnect is blocking... :)

Another thing i want to do is pulsate the light while a wifi connection is being made. Those are 2 reason why having it blocking it really inconvenient.

I could.... use interrupts and work around it that way, but that feels a bit hacky.

So therefore i'm wondering if it would be possible to make autoconnect non blocking? 

An approach that i can think of would be to let WiFiManager know that i want to use auto connect (say with a new function: setAutoconnect(true)). Then have another new function like loop() or handle() that should be placed in the main loop function to handle the wifi connection as it's made.

Just curious...

Cheers,
Mark

tablatronix, 2019-07-13:
#379 
#818 

## Development branch
#500

closed 2019-07-13 by tablatronix

markg85, 2019-07-13:
Ha, that's awesome!
I am hesitant to upgrade to the dev branch, but i think i'm going to give it a shot now. As with this it already has 3 features that i requested before...