tzapu/WiFiManager #913: Non blocking mode is still very much blocking

markg85, 2019-07-13:
Hi,

I'm on the development branch now :)

So i did set "setConfigPortalBlocking" to false and followed that with a process() call in my loop function.

But that's still blocking.
I'm guessing that only make it non blocking in the **config portal**, aka when there is no connection or it failed to connect. It's still very much blocking at the point where it tries to make a connection. 

The function:
uint8_t WiFiManager::connectWifi(String ssid, String pass)

is the one that triggers waiting. It has this if/else statement:
```
  if (ssid != "") {
    wifiConnectNew(ssid,pass);
    if(_saveTimeout > 0){
      connRes = waitForConnectResult(_saveTimeout); // use default save timeout for saves to prevent bugs in esp->waitforconnectresult loop
    }  
    else {
       connRes = waitForConnectResult(0);
    }
  }
  else {
    // connect using saved ssid if there is one
    if (WiFi_hasAutoConnect()) {
      wifiConnectDefault();
      connRes = waitForConnectResult();
    }
    else {
      DEBUG_WM(F("No saved credentials, skipping wifi"));
    }
  }
```

And if i follow waitForConnectResult then it actually waits (busy waiting loop) till it has a connection. That's the issue. 

I'm looking at how to hack that to be not blocking, to be continued.

Cheers,
Mark

tablatronix, 2019-07-14:
Thats when you save the portal... not sure what you mean

tablatronix, 2019-07-14:
Do you mean autoconnect?  You can simple not use it and use your own connect wait loop

markg85, 2019-07-14:
In my code i'm calling autoconnect. I obviously want the features of WiFiManager, aka pop up an AP if it can't make a connection and connect to an existing AP if it can. The `autoconnect` function eventually calls `connectWifi` where it is trying to connect to an existing AP if no user/pass is given (if both are emtpy). `connectWifi` is eventually* calling `waitForConnectResult` which blocks.

I don't think i can use my own connect in my main loop as i would lose the setup part that i also want if i can't make a connection. Unless you tell me otherwise that is :)

I did actually get it to work but it requires quite a different handling of `connectWifi` and `waitForConnectResult`. In fact, `waitForConnectResult` has to be rewritten mostly and put in the `process` method. I do have much of that now but i'm quite sure i currently broke the setup part... So that's definitely not something i'm going to share now.

\* unless it has no credentials, but that defeats the point entirely as i would have no wifi connection.

tablatronix, 2019-07-14:
You just startconfigportal when you need it..

markg85, 2019-07-14:
Then how do i make a connection to Wifi?

tablatronix, 2019-07-14:
The esp automatically connects to wifi or you can call wifi.begin

kbickham, 2019-08-14:
It's not remotely blocking in the way the original library did, with that my display would not sync with my controller at all.  Dev kit C  on a prototype for a dehyrdator machine with a bunch of sensors. This whole deal is fairly timing sensitive...just moved from a mega1280 this weekend. It's syncing fine, portal is popping up on it's own. I have some indicators running off of if (WiFi.status() == WL_CONNECTED) that didn't go on, but I need to go read the debug uart port so  can get iron this out.

tablatronix, 2019-08-14:
I really suggest not using autoconnect, I would require user long press or special boot sequence to enter config mode for a project like that