tzapu/WiFiManager #914: ESP does not re-connect to the wifi network with the stored parameters

Portia-Lin, 2019-07-14:
Hello!

I have a problem with automatic connection to the wifi network after ESP8266 restart.
I can`t solve this problem. 

I'm using a standard sketch "AutoConnectWithReset".
https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnectWithReset

After pressing the reset button, ESP8266 does not connect into a choosen network. It was start with the AP mode.
And when I connect to an access point again there writes this:

[image]

A similar problem was here: https://github.com/tzapu/WiFiManager/issues/91
But those solutions did not help.
I don't know why this is happening. Can someone help me in solving this problem?

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master

**Esp8266/Esp32:**
- [ ] ESP8266

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP12 F

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size:                "4M(1M SPIFFS)"

### Debug Messages

Debug messages is here:
https://pastebin.com/JJj7at2j

Portia-Lin, 2019-07-17:
The problem was in the library. I downloaded another version from the internet and everything worked. But why so ... In that case, I used the latest version of the library (master).
Now I used this [Library](http://s000.tinyupload.com/index.php?file_id=24506640731354394613)

closed 2019-07-17 by Portia-Lin

tablatronix, 2019-07-17:
Try hotfixes branch or development , also erase flash