tzapu/WiFiManager #916: HTML injection freaking out

OoM-JaN, 2019-07-23:
Hi 

Great piece of work. It's a great library but I'm stuck in the HTML insert.
Here is my code (partially):

```
String HtmlPage(){
    String htmlPageNoWiFi =
     String ( "<H1>TEST</H1>");
    return htmlPageNoWiFi;
}

void setup(){

  String y = HtmlPage();
  const char* z = "<H1>TEST</H1>";
  WiFiManagerParameter custom_text(z);

  wifiManager.addParameter(&custom_text);
  Serial.println(y);
  Serial.println(z);
```

This works perfectly and both 'y' and 'z' give the same output on the serial monitor.
If I change   WiFiManagerParameter custom_text(z); to   WiFiManagerParameter custom_text (y); I receive an error:
> no matching function for call to 'WiFiManagerParameter::WiFiManagerParameter(String&)'

What am I doing wrong or how can I change (y) to work within the WifiManager?

tablatronix, 2019-07-23:
convert to const char* ?

OoM-JaN, 2019-07-23:
const char* is the only way that custom_text 'eats' the String.
`WiFiManagerParameter custom_text("<H1>TEST</H1>") `
Works as well.

```
String z = "<H1>TEST</H1>";
'WiFiManagerParameter custom_text(z);
```
Doesn't work

tablatronix, 2019-07-23:
Yeah you are trying to pass a string in, there is no string handler, maybe we should add one

wvsbsp, 2019-08-12:
does ths work?
String z = "<H1>TEST</H1>";
'WiFiManagerParameter custom_text(z.c_str());