tzapu/WiFiManager #918: Storing SSID, IP

esp8, 2019-07-30:
#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] staging (master/dev) 2.5.0

### Description

I am playing with example AutoConnectWithStaticIP.ino . I changed only IP address. The problem is with storing parameters . When I try to save new ssid informations,  It seems that the parameters are not written in to the memory.
Example AutoConnect.ino works OK. If I upload AutoConnectWithStaticIP.ino after storing ssid and password with AutoConnect.ino  example (and not erase the memory) than esp connects to my network, but the parameters are fixed. It is not possible to change values. On the log output it is all normal. Any idea?

### Debug Messages

```
*WM: WiFi save
*WM: static ip
*WM: 192.168.1.18
*WM: static gateway
*WM: 192.168.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.18
*WM: Already connected. Bailing out.
connected...yeey :)
local ip
192.168.1.18
*WM: freeing allocated params!
```

tablatronix, 2019-07-30:
parameters are not stored anywhere, you have to save them yourself.

esp8, 2019-07-31:
hmm. 
In example AutoConnect.ino when I am in AP mode and save SSID and password, these parameters are stored somewhere, because after cycling power esp connects to network.

If i do the same with AutoConnectWithStaticIP.ino SSID and password is lost.
I tried on 3 nodemcu v1.0 and the situation is always the same.

tablatronix, 2019-07-31:
oh ok, hmm well esp does the saving for wifi creds, might be something wrong with those examples, you these both tested on the same hardware? Sometimes flash needs to be erased as it gets corrupted

wvsbsp, 2019-08-11:
I'm currently developing a project, where i am saving SSID, password, mqtt-server and topic into a config-file in the SPIFFS. I had the problem of getting the SSID and Password out of the WifiManager.

My solution to this was to edit the wifimanager.cpp, Lines 350 ff.
Here two getter-functions are commented out, which are String WiFiManager::getSSID() and String WiFiManager::getPassword().
Also the if-clause mae problems, so my getter-function just lokks like this:
` String WiFiManager::getSSID() {
  /*if (_ssid == "") {
    DEBUG_WM(F("Reading SSID"));
    _ssid = WiFi.SSID();  <<<<=== this seems to be a problem in the original code
    DEBUG_WM(F("SSID: "));
    DEBUG_WM(_ssid);
  }*/
  return _ssid;
  }`

Then add the prototypes in wifimanager.h and you're done. In my project i'm reading the custom values and SSID and password like this:
`    strcpy(mqtt_server, custom_mqtt_server.getValue());
    strcpy(mqtt_port, custom_mqtt_port.getValue());
    strcpy(mqtt_topic, custom_mqtt_topic.getValue());
    // SSID and Passwort from WifiManager 
    String sTemp=wifiManager.getSSID();
    if(sTemp != "") strcpy(ssid, sTemp.c_str()); // don't overwrite with empty sting
    sTemp=wifiManager.getPassword();
    if(sTemp != "") strcpy(pass, sTemp.c_str()); // don't overwrite with empty sting
    Serial.println("New values: ");
    Serial.println(ssid);
    Serial.println(pass);
    Serial.println(mqtt_server);
    Serial.println(mqtt_topic);
    // save configuration in SPIFFS
    saveConfig();`

I would like to see the getter-functions reactivated in the next version of WifiManager (which is a superb library!)

tablatronix, 2019-08-11:
The esp already provides these

wvsbsp, 2019-08-11:
> The esp already provides these
what do you mean? I'm currently using WifiManager Version 0.14.0 and i had to remove the comment and add prototypes to make the getter-functions working. Is it meant to be like that or is there a newer version of WifiManager with those functions enabled?

tablatronix, 2019-08-12:
the ESP stores credentials itself, wifimanager does not handle it.

looks in development branch, i added a helper getter

```C++

String WiFiManager::WiFi_SSID(){
  #ifdef ESP8266
    return WiFi.SSID();
  #elif defined(ESP32)
    //@todo , workaround only when not connected
    wifi_config_t conf;
    esp_wifi_get_config(WIFI_IF_STA, &conf);
    return String(reinterpret_cast<const char*>(conf.sta.ssid));
  #endif
}

```

tablatronix, 2019-08-12:
https://github.com/tzapu/WiFiManager/issues/887