tzapu/WiFiManager #920: Is it possible to stop the AP on successful connection?

Alives, 2019-08-03:
I'd like to turn off the AP mode once there is a successful wifi connection.  I tried stopWebPortal() but that probably just stops the http server.  Is it possible to also turn off the AP?

Thanks

Alives, 2019-08-06:
WiFi.softAPdisconnect(true) seems to do it properly.

closed 2019-08-06 by Alives

dknop97, 2019-08-08:
I have tried to use the WiFi.softAPdisconnect(true) in my code, but don't worked. So, I added this line in the library (WiFiManager.cpp file -> line 276, in WiFiManager::connectWiFi..., before the first if condition, and it's worked for me. 
Sorry if I make somes mistakes, I'm improving my English yet.

tablatronix, 2019-08-08:
```
    //manually stop the config portal if started manually, stop immediatly if non blocking, flag abort if blocking
    bool          stopConfigPortal();
```