tzapu/WiFiManager #921: Pass config url to persistent http server?

Alives, 2019-08-03:
Using development branch at 34d9a975b5339bb5510f5718fbd1530dff9de733

Once wifi is connected, I'd like to still be able to access the configuration menu to change stored credentials and sa "/update" or something.  Would it be possible to have an ESP8266WebServer.h .on() configuration to pass any clients accessing "/update" to WiFiManager to change configs?

Thanks

tablatronix, 2019-08-03:
There is an ondemand webportal example.

Alives, 2019-08-07:
It seems in that example that an event triggers the AP to enable as if autoConnect failed.  I'd like to be able to use a custom URI to access the WiFimanager config page to be able to modify my WiFiManagerParameters on the fly without running the whole AP to do so, using a URI on an existing instance of ESP8266WebServer.  I know this is probably nontrivial.  Maybe just running WiFiManager config server on a different port instead?

tablatronix, 2019-08-07:
https://github.com/tzapu/WiFiManager/blob/development/examples/OnDemand/OnDemandWebPortal/onDemandWebPortal.ino

Literally what it does.

If you want advanced stuff like adding routes check the dev example

Alives, 2019-08-18:
Yes, but I can't merge my own URI handlers with that.  Can you add some HTTP server options so I can pass in my own URIs and function pointers for handling them?  Or make server public so I can access it?  Would that work?

tablatronix, 2019-08-18:
DEV/ example

```C++
wm.setWebServerCallback(bindServerCallback);

void bindServerCallback(){
  wm.server->on("/custom",handleRoute);
  // wm.server->on("/info",handleRoute); // can override wm!
}

void handleRoute(){
  Serial.println("[HTTP] handle route");
  wm.server->send(200, "text/plain", "hello from user code");
}
```

Alives, 2019-08-21:
Thanks!  I was looking at the wrong example, sorry about that.

How can I get my parameters to show up after `wm.startWebPortal()`?  I don't see anything that seems to remove them.  If I forcibly re-add them prior to `startWebPortal`, they still don't show up on the `/wifi` page.

```
wm.stopConfigPortal();
wm.addParameter(&param_blah);
wm.startWebPortal()
```

Thanks for your help.

tablatronix, 2019-08-21:
You will probably have to change the scope of wm obj.

`WiFiManager wm;` to global outside of a function or setup for example

Alives, 2019-08-24:
That solved it.  Thanks.

One last issue with this...

When using `startWebPortal()`, it starts the DNS server as well.  Can you add a function or modify the code so that the web portal can be started without DNS as well since it's not needed (in this case)?

Phil1pp, 2021-07-31:
> When using `startWebPortal()`, it starts the DNS server as well. Can you add a function or modify the code so that the web portal can be started without DNS as well since it's not needed (in this case)?

Any progress on this?

tablatronix, 2021-08-01:
#1210

Phil1pp, 2021-08-02:
Thanks.
I was using the library available in Arduino. Didn't know that it was outdated.
After pulling the latest source directly from github it was working as expected.

tablatronix, 2021-08-04:
Hmm yeah i need to cut a beta sorry about that