tzapu/WiFiManager #923: Use  Wifi Manager With Lora Without connecting to Wifi

08435051, 2019-08-12:
Hello I have a project were I'm using an esp32 TTGO(lora).
What I would like to do is open the AP, type in costume parameters (using Spiffs) and save without having to connect to any wifi.

Because the boards are Lora I do not want to connect to any networks I just want to be able to update custom settings using the AP Spiffs custom method.

IS there something I can update in the library were it will save the setting without putting in an SSID and Password.

tablatronix, 2019-08-12:
you can do this in the development branch, you can even disable the wifi menu and use setup instead for custom parameters as they are now separate from wifi

08435051, 2019-08-13:
Thank you very much Shawn that removed the Wi-Fi menu.  The last problem I seem to have is it’s not saving the custom  parameters.  If I click on setup then enter mqtt server when the device is turned off the setting are lost.

Thank you in advance .

Nathan Hanson| IT Network Manager |Kalianna School Bendigo
Nolan Street Bendigo |P: 03 54421311
Department of Education and Training
E: hanson.nathan.h@edumail.vic.gov.au<mailto:hanson.nathan.h@edumail.vic.gov.au>

IMPORTANT - This email and any attachments may be confidential. If received in error, please contact us and delete all copies. Before opening or using attachments check them for viruses and defects. Regardless of any loss, damage or consequence, whether caused by the negligence of the sender or not, resulting directly or indirectly from the use of any attached files our liability is limited to resupplying any affected attachments. Any representations or opinions expressed are those of the individual sender, and not necessarily those of the Department of Education and Training.

tablatronix, 2019-08-13:
Wm does not save anything you will have to retrieve and save in your code since storage is up to you, A param child class to save to spiffs might be cool

08435051, 2019-08-13:
Yep Spiffs would be neat.

If I add mqtt server name in setup then connect to my  wifi, SF parameters get saved.

 I added  below code to to AutoConnectWithFSParameters(which works by removing wifi)
std::vector<const char *> menu = {"param","close","sep","erase","restart","exit"};
  wm.setMenu(menu); // custom menu, pass vector

but because I don't want to connect to wifi just have FS parameters saved under setup what would I need to change in the code? At the moment the wifi is going into a blocking loop and sf parameters are not saved because no connection is made. 

 I did think about putting in  wm.setConfigPortalBlocking(false); Although that gets passed the wifi issue it then doesn't  open up the captive portal even  though AutoConnectAP shows in the wifi list.  

I'm using Mqtt as an example.  I'm going to add a field called Key were the customer will add a  custom  key that encrypts data over Lora.

tablatronix, 2019-08-13:
The parameter example show how this works and adds spiffs saving

08435051, 2019-08-15:
Thank you so much for your help

I managed to get it to save by going to configure wifi.  I left SSID and Password blank, entered information into custom parameters and hit saved, that saved without connecting to a Wi-Fi network.
 Because I don’t want to have the Wi-Fi con fig show I added

std::vector<const char *> menu = {"param","close","sep","erase","restart","exit"};
  wm.setMenu(menu); // custom menu, pass vector

This removes Wifi Config.  When I click on set up and enter custom information and hit save its not saving.  What is it that I’m doing wrong.
I have attached my code

Regards

08435051, 2019-08-15:
#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

#ifdef ESP32
  #include <SPIFFS.h>
#endif

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40];
char mqtt_port[8];

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setupSpiffs(){
  //clean FS, for testing
  // SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
        

        

        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  setupSpiffs();

  // WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;
std::vector<const char *> menu = {"param","close","sep","erase","restart","exit"};
  wm.setMenu(menu); // custom menu, pass vector
  //set config save notify callback
  wm.setSaveConfigCallback(saveConfigCallback);

  // setup custom parameters
 
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
  

  //add all your parameters here
  wm.addParameter(&custom_mqtt_server);
  wm.addParameter(&custom_mqtt_port);

  
 wm.setBreakAfterConfig(true);

  // set dark theme
  wm.setClass("invert");
 
  if (!wm.autoConnect("Node")) {
    delay(3000);
    // if we still have not connected restart and try all over again
    
  }

  
  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"]   = mqtt_port;
 

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.prettyPrintTo(Serial);
    json.printTo(configFile);
    configFile.close();
    //end save
    shouldSaveConfig = false;
   
  
  }

}

void loop() {
  // put your main code here, to run repeatedly:

}

tablatronix, 2019-08-16:
Thanks for the code, Ill make sure it is getting to the sketch first, and that there is not a bug