tzapu/WiFiManager #924: Add helper getters for esp stored ssid and password

tablatronix, 2019-08-12:
I already added one for SSID, add one for getting password from stuct.

wvsbsp, 2019-08-12:
Thanks al lot!

tablatronix, 2019-08-13:
```C++
struct station_config {
    uint8 ssid[32];
    uint8 password[64];
    uint8 bssid_set;    // Note: If bssid_set is 1, station will just connect to the router
                        // with both ssid[] and bssid[] matched. Please check about this.
    uint8 bssid[6];
    wifi_fast_scan_threshold_t threshold;
#ifdef NONOSDK3V0
    bool open_and_wep_mode_disable; // Can connect to open/wep router by default.
#endif
};

bool wifi_station_get_config(struct station_config *config);
bool wifi_station_get_config_default(struct station_config *config);
```

tablatronix, 2019-09-11:
Needs testing

## ESP8266

- - [ ] getWiFiSSID(true)
    - - [ ] Not connected
    - - [ ] Connected to default
    - - [ ] Connected to current
- - [ ] getWiFiSSID(false)
    - - [ ] Not connected
    - - [ ] Connected to default 
    - - [ ] Connected to current

- - [ ] getWiFiPass(true)
    - - [ ] Not connected
    - - [ ] Connected to default
    - - [ ] Connected to current
- - [ ] getWiFiPass(false)
    - - [ ] Not connected
    - - [ ] Connected to default 
    - - [ ] Connected to current

## ESP32

_esp32 is different, it typically only returns info if wifi is init_
SSID, so some of these will be empty strings or persistent will be ignored

- - [ ] getWiFiSSID(true)
    - - [ ] Not connected
    - - [ ] Connected to default 
    - - [ ] Connected to current
- - [ ] getWiFiSSID(false)
    - - [ ] Not connected
    - - [ ] Connected to default 
    - - [ ] Connected to current

- - [ ] getWiFipass(true)
    - - [ ] Not connected
    - - [ ] Connected to default 
    - - [ ] Connected to current
- - [ ] getWiFipass(false)
    - - [ ] Not connected
    - - [ ] Connected to default 
    - - [ ] Connected to current

wvsbsp, 2019-09-22:
I tried it in two different projects, seems do do the trick.
The only thing is, that i'm not quite shure what the argument true/false is making with the result.

tablatronix, 2019-09-22:
One reads the default flash one reads from current config, esp32 might not matter but esp8266 will give you different configs if you disable persistent and connect to a different ssid than the default it has saved, normally you would leave it blank and get default autoconnect settings, but its there for testing as esp32 might not return anything if not connected, but it looks like some stuff was added to help wifi init and check status

commit 999da6d1 (2019-09-11): fixes #924 wifi info getters

gets ssid and psk with persistent param, needs testing
closed 2021-01-04 by tablatronix