tzapu/WiFiManager #925: Change template

array81, 2019-08-14:
I think I understood how edit template file and the generate HTTP_HEAD, HTTP_STYLE, ... strings.
Is there a way to pass these strings on my device code without edit WifiManager.h file?

I'd like don't change your code but I need change HTTP_HEAD_END to add imgae logo by base64 and HTTP_STYLE to change color button.

tablatronix, 2019-08-14:
I never figured out a way to predefine in sketch or include before library, I have seen it done, I just have no idea how to do it.

array81, 2019-08-20:
I think a similar feature should be added.
You should not have to change the library for this kind of thing and it is inconvenient if one has multiple projects with different templates.

tablatronix, 2019-08-20:
you can override in build flags 
`#ifndef WIFI_MANAGER_OVERRIDE_STRINGS`

and set your own file however you want

tablatronix, 2019-08-20:
There is an issue to create a few hooks for basic stuff. Ill check where we left off on the header hook

tablatronix, 2019-08-20:
you know you can inject in head right ?
```
    //if this is set, customise style
    void          setCustomHeadElement(const char* element);
```

a callback might be better, but for now this might help you

array81, 2019-08-20:
On version 0.14 I not found WIFI_MANAGER_OVERRIDE_STRINGS reference, what is it?

tablatronix, 2019-08-20:
Development branch

array81, 2019-08-21:
In development version I can see this warning on "strings_en.h":
`!!! THIS DOES NOT WORK, you cannot define in a sketch, if anyone one knows how to order includes to be able to do this help!`

Are you sure that WIFI_MANAGER_OVERRIDE_STRINGS works?

tablatronix, 2019-08-21:
It has not been tested, and you cannot define in sketch but you can define in build flags, no idea if arduino lets you do that or not, but platformio does.