tzapu/WiFiManager #926: esp32: Captive Portal doesnot work on windows 10

amrlsayed, 2019-08-15:
### Basic Infos
#### Hardware
ESP32 WROOM
**WiFimanager Branch/Release:**
- [ ] Master
- Development

**Esp8266/Esp32:**
- [ ] ESP8266
- ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- Other (ESP32 Wroom)

**ESP32 Core Version:	
-  1.0.2 ( Downloaded from Arduino Board Manager)

### Description
I have a more complex project. but to make it simple I am testing only the "OnDemandConfigPortal" Example. It works Good on Android but not on Windows 10. I tried disabling the firewall without luck.
when I connect to the AP from the PC I see this debug message more frequently
```
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
```
I think those are not errors because result 0 is the LWIP ERR_OK
sometimes when i connect to the AP. it redirects me to msn.com but then when i type the IP (192.168.4.1) on the browser I get no response "ERR_CONNECTION_TIMEOUT" then the output in the serial terminal is 
```
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> wpad.mt.local 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
```
when I type ipconfig on windows cmd. I find that it gets an IP from the AP
```
Wireless LAN adapter Wi-Fi:
   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::1c0c:5a92:c42f:6a4e%3
   IPv4 Address. . . . . . . . . . . : 192.168.4.2
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : 192.168.4.1
```
### Settings in IDE

Module:  		ESP32 Dev Module

Additional libraries: 
    no additional libraries
### Sketch

```cpp

/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

int timeout = 120; // seconds to run for

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW) {
    WiFiManager wm;    

    //reset settings - for testing
    //wifiManager.resetSettings();
  
    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);

    if (!wm.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.restart();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");

  }

  // put your main code here, to run repeatedly:
}
```

### Debug Messages

```
WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OnDemandAP
*WM: [2] AP has anonymous access! 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> wpad.mt.local 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 

```

viperk1, 2019-10-03:
I'm running into the same issue. Hope this can be diagnosed!

tablatronix, 2019-10-03:
check firewall

amrlsayed, 2019-10-05:
Hi Shawn,
I disabled the firewall but same problem.
Do you mean it's working good with you on windows and the problem would be our environment such as Firewall ?

tablatronix, 2019-10-05:
No, windows 10 firewall can block captive portals

connornishijima, 2019-12-04:
It doesn't work on a Galaxy S8 running Android 9 (Pie)

iw2lsi, 2021-05-27:
same problem here... on Windows 10 the captive portal is not opened... and the connection is not established; seems to works on iOS.

tablatronix, 2021-05-27:
Try without firewall ?

iw2lsi, 2021-05-27:
Hi,

   yes... firewall was not enabled...

       Giampaolo

tablatronix, 2021-05-27:
It has been awhile since I last used windows 10, have you disabled other network connections ?

iw2lsi, 2021-05-27:
Hi Shawn,

   that's a good point... I have a wired network too... I'm checking it
right now...

        Giampaolo

iw2lsi, 2021-05-27:
Hi...

   no... it does not work even with the wired card disabled... this is the
log... could it just be a timeout problem ?

         Giampaolo

 Starting
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter:
*WM: [1] AutoConnect
*WM: [1] No Credentials are Saved, skipping connect
*WM: [2] Starting Config Portal
*WM: [2] AccessPoint set password is VALID
*WM: [1] password
*WM: [3] WIFI station disconnect
*WM: [3] WiFi station enable
*WM: [2] Disabling STA[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 -
STA_STOP
[2] Enabling AP
*WM: [1] StartAP with SSID:  AutoConnectAP
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
*WM: [1] SoftAP Configuration
*WM: [1] --------------------
*WM: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
*WM: [1] password:         password
*WM: [1] ssid_len:         13
*WM: [1] channel:          1
*WM: [1] authmode:
*WM: [1] ssid_hidden:
*WM: [1] max_connection:   4
*WM: [1] country:          CN

*WM: [1] beacon_interval:  100(ms)
*WM: [1] --------------------
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
*WM: [2] WiFi Scan completed in 5304 ms
*WM: [2] Config Portal Running, blocking, waiting for clients...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
*WM: [2] Portal Timeout In 3 seconds
*WM: [1] config portal has timed out
*WM: [3] configportal abort
*WM: [2] disconnect configportal
*WM: [0] [ERROR] disconnect configportal - softAPdisconnect FAILED
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_DISCONNECTED
*WM: [2] wifi mode: STA
*WM: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 15 - AP_STOP
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 15 - AP_STOP