tzapu/WiFiManager #931: Exception 3  WiFiManager::getParamOut() when ConfigPortal started

dontsovcmc, 2019-09-10:
### Basic Infos
I try the last ESP8266 SDK with development branch and it's falls when ConfigPortal started.
How I can debug this problem? It's important to use latest SDK =(. Thanks!

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] nodemcu
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
don't know

### Description
I try development branch with 2.5.2 SDK and it fails. I'm confuzed, cause it works with 2.0.4 version...
2.1.0, 2.2.0 also raise exceptions.

I use ~30 params in config. 
```
*WM: [3] re-allocating params bytes: 100
*WM: [2] Added Parameter:
*WM: [2] Added Parameter:
*WM: [2] Added Parameter:
*WM: [2] Added Parameter: ch0
*WM: [2] Added Parameter:
*WM: [3] Updated _max_params: 30
*WM: [3] re-allocating params bytes: 120
*WM: [2] Added Parameter:
000:00:00:00:319  NOTICE    (AP) : start config portal
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  Waterius_0.9.0
*WM: [2] AP has anonymous access! 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 2192 ms
*WM: [2] WiFi Scan ASYNC found: 9
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [2] Scan is cached 3409 ms ago
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 4739 ms ago
*WM: [1] 9 networks found
*WM: [2] AP:  -48 doddd
*WM: [2] AP:  -53 dodd
*WM: [2] AP:  -64 mgts201
*WM: [2] AP:  -64 MGTS_GPON_58DA
*WM: [2] AP:  -70 romper-stomper
*WM: [2] AP:  -73 Onlime_205
*WM: [2] AP:  -74 MGTS_GPON_1235
*WM: [2] AP:  -85 MGTS_GPON_233
*WM: [2] AP:  -86 Beeline_2G_F24152
```

### Versions
```
*WM: [1] getCoreVersion():          2_5_2
*WM: [1] system_get_sdk_version():  2.2.1(cfd48f3)
*WM: [1] system_get_boot_version(): 31
*WM: [1] getFreeHeap():             36344
```

### Exception decode

```
Decoding 48 results
0x40269e21: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4026cf5f: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4020dec2: WiFiManager::getParamOut() at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.h line 117 (discriminator 2)
:  (inlined by) WiFiManager::getParamOut() at /Users/dontsov/Documents/Arduino/libraries/WiFiManager-waterius_release_090/WiFiManager.cpp line 1285 (discriminator 2)
0x40217c6c: String::changeBuffer(unsigned int) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 179
0x4020daf7: WiFiManager::getStaticOut() at /Users/dontsov/Documents/Arduino/libraries/WiFiManager-waterius_release_090/WiFiManager.cpp line 1222
0x4026cf14: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4026cf55: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40218359: String::concat(__FlashStringHelper const*) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.h line 268
:  (inlined by) String::concat(__FlashStringHelper const*) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 396
0x4026b4c6: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40213344: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 37
0x4026b4c6: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40210067: WiFiManager::handleWifi(bool) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.h line 117
:  (inlined by) WiFiManager::handleWifi(bool) at /Users/dontsov/Documents/Arduino/libraries/WiFiManager-waterius_release_090/WiFiManager.cpp line 973
0x4021cc74: std::_Function_handler    (WiFiManager*, bool)> >::_M_invoke(std::_Any_data const&) at /Users/dontsov/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x40212748: FunctionRequestHandler::canHandle(HTTPMethod, String) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 20
0x401000a9: std::function ::operator()() const at /Users/dontsov/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/functional line 2465
0x4021337a: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 44
0x40217f40: String::String(String const&) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.cpp line 41
0x40213405: ESP8266WebServer::_handleRequest() at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 599
0x4021361c: ESP8266WebServer::handleClient() at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 308
0x4021c50b: WiFiUDP::parsePacket() at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiUdp.cpp line 199
0x4021503c: DNSServer::processNextRequest() at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/DNSServer/src/DNSServer.cpp line 166
0x40211d15: WiFiManager::processConfigPortal() at /Users/dontsov/Documents/Arduino/libraries/WiFiManager-waterius_release_090/WiFiManager.cpp line 610
0x40210ce8: WiFiManager::configPortalHasTimeout() at /Users/dontsov/Documents/Arduino/libraries/WiFiManager-waterius_release_090/WiFiManager.cpp line 450
0x4026b26c: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40218ff0: esp_yield at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 91
0x40212071: WiFiManager::startConfigPortal(char const*, char const*) at /Users/dontsov/Documents/Arduino/libraries/WiFiManager-waterius_release_090/WiFiManager.cpp line 577
0x402173c9: Print::write(char const*) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.h line 60
0x40217424: Print::print(char const*) at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.cpp line 122
0x402075d0: setup_ap(Settings&, SlaveData const&, CalculatedData const&) at /var/folders/6d/w28yfrld7b7c_v7ss8cngjb40000gn/T/arduino_build_501521/sketch/setup_ap.cpp line 210
0x4026a06c: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269e21: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269d48: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269d96: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4026a06c: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269fae: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269e70: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269ebe: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269ab9: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40269ceb: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4026a088: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4026a000: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4026a00e: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4026a020: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40206b30: operator   at /var/folders/6d/w28yfrld7b7c_v7ss8cngjb40000gn/T/arduino_build_501521/sketch/Logging.h line 11
:  (inlined by) MasterI2C::getSlaveData(SlaveData&) at /var/folders/6d/w28yfrld7b7c_v7ss8cngjb40000gn/T/arduino_build_501521/sketch/master_i2c.cpp line 106
0x4020584c: loop at /Users/dontsov/Documents/CODE/waterius/ESP8266/main/main.ino line 73
0x40203421: setup at /Users/dontsov/Documents/CODE/waterius/ESP8266/main/main.ino line 32
0x402190a0: loop_wrapper() at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 125
0x40101125: cont_wrapper at /Users/dontsov/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/cont.S line 81
```

dontsovcmc, 2019-09-10:
I try to debug and found the reason in Custom Parameters! It's good news.

tablatronix, 2019-09-10:
if you set DEBUG_MAX in .h file for debuglevel
`    uint8_t       _debugLevel         = DEBUG_MAX;`

you will get debugging for heap size in logs.
I am guessing you are out of memory, and sdk uses more now ?

dontsovcmc, 2019-09-10:
```
*WM: <- page string len  6317
MEM: 36368
*WM: <- page string len  6394
MEM: 36256
*WM: <- page string len  6475

Exception (3):
epc1=0x40213214 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402469f5 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff9d0 end: 3fffffc0 offset: 01a0
3ffffb70:  402469f5 4024a00d 3ffffbc0 3ffffc64 
```

tablatronix, 2019-09-10:
have you tried it with 2.04? to see a difference?

dontsovcmc, 2019-09-10:
@tablatronix there is no much difference in memory =(. I try to check heap segmentation.. may be there is another measurements?

dontsovcmc, 2019-09-13:
@tablatronix do you know about this https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html ? I use PROGMEM string with WifiManager, that don't have _P() functions to works with this type of strings... I check it

tablatronix, 2019-09-13:
all html strings are in progmem, I have no idea how you that would work with non constants

dontsovcmc, 2019-09-13:
IT WORKS! Without PROGMEM string for parameters...
I think, somewhere wrong function for Progmem strings...
p.s. also https://forums.adafruit.com/viewtopic.php?f=56&t=106457#wrap

closed 2019-09-13 by dontsovcmc

tablatronix, 2019-09-13:
what did you do , put your own sketch strings in progmem ?

tablatronix, 2019-09-13:
It looks like the webserver already has this, I wonder if it has a send buffer and we can just write to it then send instread of concatenating our own string..

I have not worked much with the webserver.

ESP8266WebServer::send_P

tablatronix, 2019-09-13:
It also has methods with params for code and content_type, I wonder if we can get rid of sending our own headers now.
`
  server->sendHeader(FPSTR(HTTP_HEAD_CL), String(page.length()));
  server->send(200, FPSTR(HTTP_HEAD_CT), page);
`

tablatronix, 2019-09-13:
Ok no buffer, so is it as easy as using PSTR() ?

dontsovcmc, 2019-09-13:
Now I make "easy" changes - remove "PROGMEM" everywhere. But, I think, It better understand behaviour for WifiManager.

Maybe incorrect use const char * in functions arguments?
Like this: https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/setup_ap.h#L20

All my strings are custom HTML:
https://github.com/dontsovcmc/waterius/blob/master/ESP8266/src/setup_ap.cpp#L126

i see this patch and think my code wrong:
https://github.com/esp8266/Arduino/issues/4323

This is what I need: const char * footer = String(F("\r\n")).c_str();

tablatronix, 2019-09-13:
ok so how did you fix it ???
remove progmem ?
Can you use F() strings?

dontsovcmc, 2019-09-13:
yes, remove PROGMEM in my code. I try it today night. Write here result

dontsovcmc, 2019-09-13:
Show nothing on the screen:
```
const char * SMTH = String(F("<text>")).c_str();
WiFiManagerParameter param(SMTH);
wm.addParameter( &param);
```
Crash:
```
FlashStringParameter param(F("<text>"));
wm.addParameter( &param);
......
class FlashStringParameter : public WiFiManagerParameter {
public:
    String str;
    FlashStringParameter(const __FlashStringHelper *s)
    {
        str = s;
        setCustomHTML(str.c_str());
    }
};
```

OK:
```
String label_str(F("<text>"));
WiFiManagerParameter label_cold_info(label_str.c_str());
wm.addParameter( &label_cold_info);
```
and also:
```
WiFiManagerParameter label_("<text>");
wm.addParameter( &label_);
```

dontsovcmc, 2019-09-14:
It's interesting, that WiFiManager fails if 
1.
`build_flags = -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE  -DDEBUG_ESP_WIFI`
also with `build_flags = -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE`
2. this type of strings:
```
WiFiManagerParameter label_("<text>");
wm.addParameter( &label_); 
```

```
SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
000:00:00:00:067  NOTICE    (ESP) : Booted
wifi evt: 2
1 000:00:00:00:072  NOTICE    (I2C) : mode=1
9 3 6D C 0 0 1D 0 0 0 14 0 0 0 1E 0 0 0 000:00:00:00:088  NOTICE    (I2C) : version: 9
000:00:00:00:088  NOTICE    (I2C) : service: 3
000:00:00:00:089  NOTICE    (I2C) : voltage: 3181
000:00:00:00:090  NOTICE    (I2C) : resets: 29
000:00:00:00:094  NOTICE    (I2C) : state0: 0
000:00:00:00:098  NOTICE    (I2C) : state1: 0
000:00:00:00:102  NOTICE    (I2C) : impulses0: 20
000:00:00:00:107  NOTICE    (I2C) : impulses1: 30
000:00:00:00:112  NOTICE    (CFG) : CRC ok
000:00:00:00:115  NOTICE    (CFG) : WATERIUS
000:00:00:00:119  NOTICE    (CFG) : email=
000:00:00:00:124  NOTICE    (CFG) : host=https://cloud.waterius.ru key=12
000:00:00:00:134  NOTICE    (CFG) : BLYNK.CC
000:00:00:00:138  NOTICE    (CFG) : host=blynk-cloud.com key=
000:00:00:00:143  NOTICE    (CFG) : email=
000:00:00:00:147  NOTICE    (CFG) : MQTT
000:00:00:00:150  NOTICE    (CFG) : host= port=1883
000:00:00:00:155  NOTICE    (CFG) : login= pass=
000:00:00:00:159  NOTICE    (CFG) : topic=waterius/12/
000:00:00:00:165  NOTICE    (CFG) : COUNTERS
000:00:00:00:169  NOTICE    (CFG) : channel0_start=0.00, impulses0_start=20, factor=10
000:00:00:00:176  NOTICE    (CFG) : channel1_start=0.00, impulses1_start=30
000:00:00:00:183  NOTICE    (ESP) : new impulses=20 30
000:00:00:00:188  NOTICE    (ESP) : new value0=0.00 value1=0.00
000:00:00:00:193  NOTICE    (ESP) : delta0=0 delta1=0
data  : 0x3ffe8000 ~ 0x3ffe856c, len: 1388
rodata: 0x3ffe8570 ~ 0x3ffea4ac, len: 7996
bss   : 0x3ffea4b0 ~ 0x3fff1038, len: 27528
heap  : 0x3fff1038 ~ 0x3fffc000, len: 45000
*WM: [1] getCoreVersion():          2_5_2
*WM: [1] system_get_sdk_version():  2.2.1(cfd48f3)
*WM: [1] system_get_boot_version(): 31
*WM: [1] getFreeHeap():             36688
000:00:00:00:229  NOTICE    (AP) : User requested captive portal
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter:
*WM: [2] Added Parameter: wmail
*WM: [2] Added Parameter:
*WM: [2] Added Parameter:
*WM: [2] Added Parameter:
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: ch1
*WM: [2] Added Parameter:
*WM: [2] Added Parameter:
*WM: [2] Added Parameter:
*WM: [2] Added Parameter: ch0
*WM: [3] Updated _max_params: 15
*WM: [3] re-allocating params bytes: 60
*WM: [2] Added Parameter:
*WM: [2] Added Parameter:
000:00:00:00:280  NOTICE    (AP) : start config portal
*WM: [3] WIFI station disconnect 
scandone
*WM: [3] WiFi station enable 
[... 52 lines cut ...]
0x40255065: esf_buf_alloc at ?? line ?
0x402477cd: ieee80211_beacon_alloc at ?? line ?
0x402557b3: ic_bss_info_update at ?? line ?
0x401056e4: ets_timer_arm_new at ?? line ?
0x40242dd9: ieee80211_hostap_attach at ?? line ?
0x40242e36: ieee80211_hostap_attach at ?? line ?
0x40100da0: pvPortZalloc at ?? line ?
0x40218861: calloc at ?? line ?
0x40100da0: pvPortZalloc at ?? line ?
0x4025a64e: wifi_set_broadcast_if at ?? line ?
0x401056e4: ets_timer_arm_new at ?? line ?
0x402162c0: esp_yield at ?? line ?
0x40217137: delay at ?? line ?
0x4021187a: WiFiManager::startAP() at ?? line ?
0x402692b0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4025b594: wifi_station_disconnect at ?? line ?
0x40100ac2: millis at ?? line ?
0x402122b0: WiFiManager::startConfigPortal(char const*, char const*) at ?? line ?
0x4021466d: Print::write(char const*) at ?? line ?
0x40214790: Print::print(char const*) at ?? line ?
0x40205084: setup_ap(Settings&, SlaveData const&, CalculatedData const&) at ?? line ?
0x402038b0: loop at ?? line ?
0x4020334d: setup at ?? line ?
0x40216370: loop_wrapper() at core_esp8266_main.cpp line ?
```

dontsovcmc, 2019-09-14:
oups... i think it's look like this bug: https://github.com/esp8266/Arduino/issues/4372#issuecomment-411895439

dontsovcmc, 2019-09-14:
I can't believe, that another bug ( I think it was in SSLBear) is relevant to WiFi.mode() function!!!
HTTPS request raised exception after ConfigPortal ended.
Now I found the way when it works!

```
WiFi.mode(WIFI_OFF);
WiFi.mode(WIFI_AP);
ConfigPortal()
WiFi.mode(WIFI_OFF);
WiFi.mode(WIFI_STA);
HTTPS request
```

It's f&cking terrible, that Arduino framework can't change modes correctly... I'm confused

tablatronix, 2019-09-14:
very interesting

dontsovcmc, 2019-09-15:
> very interesting

I think we should add remark to listing or documentation, that using PROGMEM strings with WifiManagerParameter is prohibited. 
I can't understand how to add support.
I think it's not necessary cause a little number of cases such this. So remark is best way.

tablatronix, 2019-09-16:
Yeah I am not entirely sure why, not sure I fully understand whats going on

dontsovcmc, 2019-09-16:
@tablatronix I try to describe: 
in esp PROGMEM string must be used only by special [fuctions](https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html): memcmp_P, strlen_P and etc.

Working with PROGMEM strings with common functions "strlen", "memcmp", "scanf" is dangerous (read "prohibited"). 

What's why this "operator=" can raise exception:
https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L543

Look at String contructors:
```
String::String(const char *cstr) {  //for common strings
    init();
    if(cstr)
        copy(cstr, strlen(cstr));
}

String::String(const __FlashStringHelper *pstr) { //for PROGMEM
    init();
    *this = pstr; // see operator =
}

String & String::operator = (const __FlashStringHelper *pstr)
{
    if (pstr) copy(pstr, strlen_P((PGM_P)pstr));  //WOW strlen_P not strlen 
    else invalidate();
    return *this;
}
String & String::copy(const __FlashStringHelper *pstr, unsigned int length) {
    if (!reserve(length)) {
        invalidate();
        return *this;
    }
    setLen(length);
    strcpy_P(wbuffer(), (PGM_P)pstr);  //WOW #2 strcpy_P not strcpy
    return *this;
}
```

tablatronix, 2019-09-16:
ok, so we would have to add some checking to avoid exceptions, a note for now will work.

dontsovcmc, 2019-09-16:
@tablatronix I can't understand how to check it =(. Only add functions with `const __FlashStringHelper *`.