tzapu/WiFiManager #933: .15.0 beta fails during compilation

Daemach, 2019-09-11:
I downloaded the .15.0 beta today and tried building but get the following error in PIO - this didn't happen before the update:

```
Compiling .pio\build\upstream_develop\src\THCG.cpp.o
In file included from src\defines.h:28:0,
                 from src\THCG.cpp:1:
lib/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

Compiling .pio\build\upstream_develop\lib114\WiFiManager\WiFiManager.cpp.o
compilation terminated.
In file included from lib\WiFiManager\WiFiManager.cpp:13:0:
lib\WiFiManager\WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

compilation terminated.
*** [.pio\build\upstream_develop\src\THCG.cpp.o] Error 1
*** [.pio\build\upstream_develop\lib114\WiFiManager\WiFiManager.cpp.o] Error 1
======================================================== [FAILED] Took 8.55 seconds ========================================================
```

tablatronix, 2019-09-11:
What are you compiling and which board?

Daemach, 2019-09-11:
ESP32 dev board.  Custom application that worked this afternoon before making that update.  Sorry, I thought I included the ESP32 part but obviously I didn't.

tablatronix, 2019-09-11:
15.0 does not support esp32, development branch does.

closed 2019-09-11 by tablatronix