tzapu/WiFiManager #946: startLocalOnlyHotspot closed by system

iadeelzafar, 2019-09-22:
I'm using [startLocalOnlyHotspot](https://developer.android.com/reference/android/net/wifi/WifiManager#startLocalOnlyHotspot(android.net.wifi.WifiManager.LocalOnlyHotspotCallback,%20android.os.Handler)) in a project of mine that is similar to this project.

Hotspot is turned off by the system after a few minutes and I see this in the Logs.

```2019-09-05 11:12:42.087 759-3392/? I/WifiService: stopLocalOnlyHotspot uid=10184 pid=16049```

Digging more, here's what this method that is being called by the system itself does.

```
 /**
     *  Method used to inform WifiService that the LocalOnlyHotspot is no longer needed.  This
     *  method is used by WifiManager to release LocalOnlyHotspotReservations held by calling
     *  applications and removes the internal tracking for the hotspot request.  When all requesting
     *  applications are finished using the hotspot, it will be stopped and WiFi will return to the
     *  previous operational mode.
     *
     *  This method should not be called by applications.  Instead, they should call the close()
     *  method on their LocalOnlyHotspotReservation.
     */
    private void stopLocalOnlyHotspot() {
```

**So, my question is how the system decides that LocalOnlyHotspot is no longer needed? My other devices are connected to the local hotspot and a server is hosted on it. But it still gets turned off.** 

Please enlighten me.

Thanks!

iadeelzafar, 2019-09-22:
P.S: I'm using the `startLocalOnlyHotspot` on a foreground service but the hotspot is closed by the system while the service is still running.

tablatronix, 2019-09-22:
I have no idea what this is

tablatronix, 2019-09-22:
This is arduino esp wifimanager not android

closed 2019-09-22 by tablatronix