tzapu/WiFiManager #948: WifiManager hangs setup after successful configuration save

afagard, 2019-09-24:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
Lolin D1 mini pro

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] staging (master/dev)

### Description

On first run, with clear memory, the device starts and I can go to the configuration portal to modify the wifi credentials and save it. On save, and over serial, the device immediately spits out:

> *WM: WiFi save
> *WM: Sent wifi save page
> *WM: Connecting to new AP
> *WM: Connecting as wifi client...
> (few seconds later)
> connected with XXXX, channel 1
> dhcp client start...
> ip:10.0.0.236,mask:255.255.255.0,gw:10.0.0.1
> *WM: Connection result: 
> *WM: 3
> station: c0:ee:fb:de:08:61 leave, AID = 1
> rm 1
> bcn 0
> del if1
> pm open,type:2 0
> mode : sta(84:0d:8e:8f:4e:2e)

And stops. Then, after about a minute and a half, my own setup begins and everything goes on normally.

> pm open,type:2 0
> mode : sta(84:0d:8e:8f:4e:2e)
> [CONFIG] There was an error opening config.json for reading. This is normal on first-run startup.
> [CONFIG] Failed to read file. Using default configuration. This is normal on first-run startup.
> [WIFI] mDNS responder started
> [WIFI] TCP server started
> [DS] Locating Dallas temperature sensors...[DS] Found 2 sensor(s).
> [DS] Added temp sensor.
> [DS] Added temp sensor.
> [BME] Locating BME280...
> [BME] Found sensor.
> [IRRIGATION] Pump off.
> [IRRIGATION] Pump on.
> [IRRIGATION] Pump off.
> [IRRIGATION] Pump on.
> *WM: freeing allocated params!
> [RAINFALL] Attempting to save rainfall data...
> [RAINFALL] rainfall.txt does not exist. At startup this is expected.
> [RAINFALL] File was written.
> [IRRIGATION] Running routine...
> [RAINFALL] Tip count from memory: 0

What exactly is happening during this timeperiod? Is this normal? Because as the serial prints suggest, the device has successfully connected after the credentials were correctly entered. Without wifimanager, and using hard-coded configuration variables, there is no delay.

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

My sketch is about 1000 lines:

https://pastebin.com/Jn6eZq43

But this is my basic setup:

```
void setup(void)
{
  Serial.begin(115200);
  pinMode(WIFI_STATUS_LED, OUTPUT);
  pinMode(ERROR_LED, OUTPUT);
  pinMode(PUMP_RELAY, OUTPUT);
  digitalWrite(PUMP_RELAY, LOW);
  digitalWrite(WIFI_STATUS_LED, LOW);
  digitalWrite(ERROR_LED, LOW); // inverse for some reason...
  WiFiManager wifiManager;
  wifiManager.autoConnect("IrrigationStationWifiSetup", "XXXX");
  wifiManager.setDebugOutput(true);
  //wifiManager.setSTAStaticIPConfig(staticIP, gateway, subnet);
  if (!SPIFFS.begin()) {
    enableErrorLed();
    Serial.println("Failed to mount the file system!");
    while (1);
  }
  resetRtc();
  createErrorsArray();
  getConfig(config); // first thing to serial print see: [CONFIG] in output above
  deviceStarted = millis();
  serverDefs();
  //initWifi();
  initWifi();
  initRtc(); // must be called after wifi
  initDs();
  initBme();
  initRg();
  relayTest();
  runner.init();
  runner.addTask(t1);
  runner.addTask(t2);
  t1.enable();
  t2.enable();
  ota();
}

void initWifi() {
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  // start mdns
  if (!MDNS.begin("station")) {
    Serial.println("[WIFI] Error setting up MDNS responder!");
    while (1) {
      digitalWrite(WIFI_STATUS_LED, HIGH);
      delay(500);
      digitalWrite(WIFI_STATUS_LED, LOW);
      delay(500);
    }
  }
  Serial.println("[WIFI] mDNS responder started");
  // Start TCP (HTTP) server
  server.begin();
  Serial.println("[WIFI] TCP server started");
  // Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);
  digitalWrite(WIFI_STATUS_LED, HIGH); // signify connection started
}
```

ermaya, 2019-10-05:
Have you read the first line of the AutoConectWifi Parameters example?
Maybe it helps.
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

afagard, 2019-10-13:
> Have you read the first line of the AutoConnectWithFSParameters example?
> Maybe it helps.
> #include <FS.h> //this needs to be first, or it all crashes and burns...

But it doesn't crash... Will try it out nevertheless.

tablatronix, 2019-10-13:
I would guess this infinite loop ?
while (1);

afagard, 2019-10-13:
> I would guess this infinite loop ?
> while (1);

within the confines of an if it isn't afaik (seen it in more than a few coding examples), in any case, it doesn't crash or print anything to the serial. there are just a few things that need to happen or all else fails, and those have the while(1) statements.

tablatronix, 2019-10-13:
Those do not look like logs from development branch