tzapu/WiFiManager #949: Reset Setting including Spiffs.fomart -> ESP8266

bgersmann, 2019-09-26:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ X] Master
- [ ] Development

**Esp8266/Esp32:**
- [X ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ X] 2.5.2
- [ ] staging (master/dev)

### Description

Hi,

I'm using a local copy of the library, because I added a "Spiffs.format()" to the reset function. 
Is there any easy way to do this without the local copy? Could you maybe add it as an optional parameter to the reset function?

Thanks in advance.

tablatronix, 2019-09-26:
hmm, WM does not use Spiffs, so it would make no sense to add this

Joennuh, 2019-09-26:
Some examples DO use SPIFFS to save extra parameters. Like this one: https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino
An SPIFFS reset function would be nice but I believe it is not absolutely necessary for the core functionality of WiFiManager.

tablatronix, 2019-09-26:
examples.
I am going to try to add a define or auto detect spiffs and do it like that

tablatronix, 2019-09-26:
Ok so WiFiManager::erase(bool opt) already exists and has an optional flag

in esp32 this can erase NVS

In esp8266 I can make it erase spiffs ( if spiffs was compiled and is incuded etc )
I just have to figure out error handling and stuff

tablatronix, 2019-09-26:
I think I will just leave it seperate, so you have to call it twice for now

tablatronix, 2019-09-26:
Will have to test of course, as of now they are seperate, I can work on combining them later if they work and I can figure out error handling.

as of now

on esp8266 
`erase()` will erase wificonfig via "ESP.eraseConfig()" or sector wipe if `WM_FIXERASECONFIG`
`erase(true)` will erase spiffs ONLY

on esp32
`erase()` will erase wifi config via disconnect
`erase(true)` will erase NVS if `WM_ERASE_NVS` and `nvs_flash_h`

tablatronix, 2019-09-26:
you can test with this

```C++
  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
    return;
  }
    FSInfo info;
    SPIFFS.info(info);
    Serial.printf("Total: %u\nUsed: %u\nBlock: %u\nPage: %u\nMax open files: %u\nMax path len: %u\n",
                  info.totalBytes,
                  info.usedBytes,
                  info.blockSize,
                  info.pageSize,
                  info.maxOpenFiles,
                  info.maxPathLength
                 );
```

tablatronix, 2019-09-26:
hmm not working, let me fix it

tablatronix, 2019-09-26:
of course it wont work, doh, I cannot detect spiffs defined in sketch.. ugh C++ linker is so anoying sometimes

Ruandv, 2020-01-15:
@tablatronix is this something that I can have a look at for you? Can i maybe put it as part of an exampe?

tablatronix, 2020-01-15:
Well I was hoping to have it automatic but I do not see any way to do it ill have to add some flags or params to let users do it instead

Ruandv, 2020-01-15:
_Suggestion:_ 
Why don't you rename the`bool WiFiManager::erase(bool opt)` to `bool WiFiManager::format()` ?
then `bool WiFiManager::erase()` will still only do  `WiFi_eraseConfig` and 
`bool WiFiManager::format()` will then go and do `spiffs.format()` and call `WiFi_eraseConfig`

Ruandv, 2020-02-22:
Is there any objections to my suggestion above? If not then I'm willing to try and implement it.

tablatronix, 2020-02-22:
Yeah but that's not really the problem, the problem is this only works with spiffs enabled.

Ruandv, 2020-02-22:
Ok thanks now I understand the bigger problem.