tzapu/WiFiManager #951: Project README.md no longer updating release notes

bperrybap, 2019-09-30:
The README.md has a section down near the bottom called "Releases" which has a summary of the updates for each release.
It appears that it hasn't been updated since the 0.12 release, so there is no release information on the later releases.

It would be nice to get this section updated with the release notes for each release especially if a release is known to require a certain core version.

Joennuh, 2019-09-30:
I do agree with @bperrybap but you can also see the releasenotes on the releases page of the repository: https://github.com/tzapu/WiFiManager/releases

tablatronix, 2019-09-30:
Release info will probably be removed from it as all issues are tagged in github, who has time ... lol

tablatronix, 2019-09-30:
Fyi the development branch doesn't even have docs yet...

bperrybap, 2019-10-01:
Automation is the key. 
Once automated, it can take no extra time and likely reduce the total human time/effort to do a release.

For my Arduino libraries, I pull the git comments and automatically generate the release notes into the readme file.

Have a look at the readme for one of my libraries: https://github.com/duinoWitchery/hd44780
(down towards the bottom of the page)

I have a script that handles everything for a release.
Not only does it insert the git comments into the readme, but it also updates the version number information in several files like header files, and in the readme file and even does the commit.

So releases are super easy to do, and very consistent once it is all automated.

You can take a look at the scripts in the hd44780 library they are under the "extra" directory.

tablatronix, 2019-10-01:
I have added doc blocks to all public functions already

tablatronix, 2019-10-01:
Nice scripts!

bperrybap, 2019-10-01:
These types of automation scripts could also do something a bit smarter with the comments like a bit of selective filtering.
For example, it could look for a leading character/string in the comment to either selectively include or selectively exclude the comment into the public release note history. That way,  when a commit was done the author could  indicate whether the commit comment was part of the public release note history in the readme file.
So not all the commit comments would have to be part of the public release notes history.

Currently, for my library, I kept it simple and just include all the comments into the public release note history.