tzapu/WiFiManager #952: WifiMulti support

Bolukan, 2019-10-07:
### Basic Infos
I would like to register multiple Wifi stations (SSID's). If the device enters the configportal it should be an option to keep (max 2) existing SSID's (show SSID's).

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [x] ESP32

### Description
My use case is I move these devices between my main network and my "mancave" network, or towards a friend. I want to keep multiple SSID's present on the device. As is known esp devices support up to max 3 SSID's.

I would like to register multiple Wifi stations (SSID's). If the device enters the configportal it should be an option to keep (max 2) existing SSID's (show SSID's).

The library should support some kind of WiFiMulti. This could be implemented as a fallback scenario if the primary SSID is not available and the scan identifies SSID-2 or SSID-3. It should try to connect to them before opening an AP.

tablatronix, 2019-10-07:
Multi ssid have to be stored somewhere.

Bolukan, 2019-10-08:
Of course. I somewhere remembered the esp stores multiple ssid’s in it’s memory (and I don’t refer to EEPROM, but the system memory) as I once analyzed the system memory storage. See wifi_station_ap_number_set and related functions for the direction I was considering. I will dig into it whether this still can be used for this purpose.

tablatronix, 2019-10-08:
It does not store those.

https://github.com/esp8266/Arduino/issues/2698

Bolukan, 2019-10-08:
I’ll look into it

tablatronix, 2019-10-09:
I guess we were assuming, it would use the current wifi_station_ap_change(INDEX)

perhaps
wifi_station_ap_number_set(5,4,3,2,1) to set the location in memory?

Bolukan, 2019-10-09:
Also for other future participants:

**bool wifi_station_ap_number_set(uint8 ap_number)** sets the number of memory slots available. 5 is a common number ([example code in the NONOS library](https://github.com/espressif/ESP8266_NONOS_SDK/blob/590fd5492276b5cd9c34c3c5470272ead632be5f/examples/IoT_Demo/include/user_config.h#L81)). It is directly correlated to the number of records and the records themselves returned by the next function:
**uint8_t wifi_station_get_ap_info (struct station_configc onfig[])** retrieves number and content of the AP's.
**uint8 wifi_station_get_current_ap_id (void)** returns the current AP.

As you state, **bool wifi_station_ap_change (uint8_t current_ap_id)** changes to the specific AP (I read about it in some old bug Espressif solved, but this also confirmed the objective). But it is buggy: other AP's disappear and if another disappeared AP is chosen the ESP8266 can't connect (so this bug distorts the function of your device). After a reset the AP's are still in memory.

Then more guesses:
Somewhere in the flash the default AP at boot-up is saved. I expect **wifi_station_get_config_default** to be related.

Trick: Set persistent=false before disconnect to keep your AP in flash (otherwise the call of wifi_station_set_config(<empty>) deletes the AP (current AP? More tests needed).

The wifi_station_ap_change function seems to corrupt the internal memory buffer with the five AP's (disappearing AP's) but not the flash memory, because after a reset AP-info reappears.

Bolukan, 2019-10-14:
I propose to close this issue, as I see not enough possibilities with the current SDK. I prefer to keep support for multiwifi based on the possibilities of Espressif’s SDK, and not use SPIFFS/EEPROM for storing SSID’s. That would complicate the library.

tablatronix, 2019-10-15:
Do you think it needs more testing, or should we pressure espressif with these findings, an sdk feature should work if it says it does.

hoekbrwr, 2019-10-15:
As I had interruption of Wifi to Internet I activated my Hotspot for Internet connection to 4G data and I would like to have automatic switchover to that in Neopixelclock sync. So this would be a nice application.

Bolukan, 2019-10-15:
Ok @tablatronix , I agree with you we should report our problems to Espressif and see whether they will solve it. If we don't ask, they cannot help. 

My guess is it would help if we deliver one or more clean test cases to show the unexpected behaviour. Otherwise a little piece of the implementation could cause or influence the outcome without Espressif and you knowing how my tests went. I can prepare some scripts but need some time (busy week+weekend). 

Two parts:
If a second SSID is started (wifi_station_set_config), the old one gets saved. But the location is not logic and multiple copies appear. (shown by wifi_station_get_ap_info)
Second issue is if you choose a SSID (wifi_station_ap_change) other entries (wifi_station_get_ap_info) disappear.

As I suspect some difference between cache and flash, the tests should try both variants.

tablatronix, 2019-10-16:
Yes if you can throw a unit test togather that would be great, you can stick it on gist if you want, so it can be forked and modified

snick-m, 2022-11-17:
I'm also in need of this feature as I'm trying to build a portable device that would likely be taken from home to school and would remember both WiFis so that it can connect to whichever is available. From my looking around the connected issues, this is not yet supported in the WiFiManager library. Where may I pick up if I were to work on getting this support?

iKK001, 2023-11-26:
does anybody know if this has been implemented ? (i.e. possibility to store two (ore more) SSID/PW to be able to use the same ESP32-device on two locations (home and work) without AccessPoint showing up but connectiong at both locations) ??

snick-m, 2023-11-27:
It has not been implemented yet as per my knowledge. I did want to work on adding support for this but never got around to actually exploring it.

iKK001, 2023-11-27:
Thanks for your feedback. Same here - I am just not knowledgable enough to dare to get into the ESP32-SDK storage of SSID/PW.    There are some solutions that store additional SSID/PW's into Flash or EEPROM. But I don't think this is safe !!!!

Therefore my question if anybody knows how to add support for two SSID/PW storage with the ESP32-SDK ?

dBitech, 2024-03-14:
I too am intrested in a solution to this