tzapu/WiFiManager #954: wifi manager

santhoshnp, 2019-10-18:
Hi team,
I am using ESP8266 in that i am working with wifi manager i have downloaded the wifi library and i am using the example code of wifi manager 
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}
In this code run successfully without any error but when i connect to try to connect the SSID it is not discovering as AutoConnectAP i dont why i have followed these link 
1) http://www.martyncurrey.com/esp8266-and-the-arduino-ide-part-5-adding-wifimanager/
2) https://randomnerdtutorials.com/wifimanager-with-esp8266-autoconnect-custom-parameter-and-manage-your-ssid-and-password/
3)https://github.com/tzapu/WiFiManager
i dont why can you please help me

tablatronix, 2019-10-18:
You do not see the access point?

lbussy, 2019-10-23:
What shows in your serial log?

belazelkos, 2019-11-27:
Hi,
I have problem compiling WiFiManager for Wemos D1, even when I try to compile example for WiFiManager.

This is error message what I get: 

In file included from C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,

                 from C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

                 from C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/Arduino.h:32,

                 from sketch\AutoConnect.ino.cpp:1:

C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))

                                                                                                                                 ^

C:\Users\B\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM'

 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                        ^

In file included from C:\Users\B\Documents\Arduino\libraries\WiFiManager\examples\AutoConnect\AutoConnect.ino:5:0:

C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

exit status 1
Error compiling for board WeMos D1 R1.

Any help with this, please!

tablatronix, 2019-11-27:
Why are you not posting your own issue for this?

webdevbrian, 2020-01-14:
@belazelkos For this error basically it boiled down to me having two versions of wifimanager installed. And the only way I found that out, was it reported I had .15 beta installed, so I uninstalled that through the library manager, then it reported that .14 was magically installed so I uninstalled that. I then from that point, installed .15 beta and it compiled and worked for me. I was getting the exact error as you. Hope this helps someone because I wasted a good 4 hours debugging this.

pipo114, 2020-04-26:
@webdevbrian Thanks for your solution ! It helps me a lot !