tzapu/WiFiManager #955: Support for DNS2?

lbussy, 2019-10-22:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [X] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [X] staging (master/dev)

### Description

I don't see support for 2 DNS servers in this lib as there is in ESP8266WiFi.  Is this somewhere I'm missing maybe?  If not, is it planned to be added?

### Settings in IDE

Module: N/A

Additional libraries: N/A

### Sketch

In `WiFiManager.h`:

```cpp
//sets config for a static IP with DNS
void setSTAStaticIPConfig(IPAddress ip, IPAddress gw, IPAddress sn, IPAddress dns);
```

tablatronix, 2019-10-22:
Was this always available?

lbussy, 2019-10-22:
> Was this always available?

It existed as far back as [2.1.0-rc1 (Jan 4, 2016 in ESP8266WiFiSTAClass::config()](https://github.com/esp8266/Arduino/blame/43a55bd404b369691fdc50a837cd586eac3198ab/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L196).  In 2.0 (Nov 30, 2015), it was still ESP8266WiFi::config() and only had the single DNS argument.