tzapu/WiFiManager #956: Hide Access point(SSID)

Muzzamil1, 2019-10-22:
Hi, Thanks for the great library!
Is it possible to hide the access point(SSID)?

Muzzamil1, 2019-10-29:
Hi @tablatronix
Do you have any idea about it?

tablatronix, 2019-10-30:
Yes it is possible I think, let me check. 

Logging into portals will be hard though

Muzzamil1, 2019-10-30:
Let me know when you check.

dknop97, 2019-10-30:
I have a trouble with the SSID, because after the success login with the captive portal, the AP SSID still detectable, but don't active, that is, I keep seeing the SSID, but I can't connect to it. Do you know something like this and a solution?

tablatronix, 2019-10-30:
That is the esp doing that, it is the default, you will need to set wifi more to sta in your code

WiFi.mode(WIFI_STA);

tablatronix, 2019-11-14:
// set ap hidden
    void          setWiFiAPHidden(bool hidden); // default false

closed 2019-11-14 by tablatronix

dknop97, 2019-11-14:
> ```
> // set ap hidden
> void          setWiFiAPHidden(bool hidden); // default false
> ```

Sorry, I'm confused. What I need to do with this? Put where? How? Sorry again.

tablatronix, 2019-11-14:
Just set that before calling autoconnect etc..

`wm.setWiFiAPHidden(true);` // hide the portal ap

tihoangyeudau, 2019-11-16:
> this is for the feature @Muzzamil1 requested
> 
> Just set that before calling autoconnect etc..
> 
> `wm.setWiFiAPHidden(true);` // hide the portal ap

when i add wm.setWiFiAPHidden(true); before autoconnect, there is access point(SSID) named : "Hidden Wifi".

i think your code is wrong.

Muzzamil1, 2019-11-18:
> this is for the feature @Muzzamil1 requested
> 
> Just set that before calling autoconnect etc..
> 
> `wm.setWiFiAPHidden(true);` // hide the portal ap

Hi @tablatronix  
Thanks, It worked.

commit a2355889 (2019-12-12): fixes #956

Not tested, not sure if captive portal works with hidden ap, test please.
adds wm.setWiFiAPHidden(bool);