tzapu/WiFiManager #957: Question:  Use of Static IP

lbussy, 2019-10-22:
Working in the development branch:

I found a reference (in a comment) in the script referring to using a static IP, and I also dug in the source code a bit:

https://github.com/tzapu/WiFiManager/blob/9a85b107fa264a5a3e2218f72b18dfab629b1260/examples/Advanced/Advanced.ino#L52

Am I correct in my understanding that to use this it should get set before calling `wm.autoConnect(ssid, pwd);` and that doing so will auto-connect to the configured access point with that IP information instead of using WiFi.config(ip, gw, sn, dns);?

I also saw:

```
wm.setShowStaticFields(true); // force show static ip fields
wm.setShowDnsFields(true);    // force show dns field always
```

The implication there is I don't have to set up custom fields to configure static IP and DNS (yay!).  What's not clear is whether those fields autosave anywhere or if I need to save them manually.  If the latter, how do I access the field values to save?  If they are auto-saved, how do I access them initially to set the setSTAStaticIPConfig() call?

I looked through all the rest of the examples as well as chased it a bit through the source code but did not find my answer.  The core documentation was likewise not helpful.  I got as far as ESP8266WiFiGenericClass::persistent() and then got lost.  Hopefully I didn't miss an obvious answer to this.

lbussy, 2019-10-23:
"Seems like" that's how it works - that `setShowStaticFields(true)` will allow setting a static IP and that `setShowDnsFields(true)` allows setting a DNS server.  Further, somehow, `ESP8266WiFiGenericClass::persistent()` saves that off automagically.  It seems like there's no need to use `setSTAStaticIPConfig()` when this is done.

Hopefully, that's correct.  It seems to work anyway.

closed 2021-06-13 by lbussy

tablatronix, 2021-06-13:
The esp will not remember static or ip info you will have to save it and restore it

lbussy, 2021-06-13:
Understood, thanks.  I was just cleaning up an old issue.