tzapu/WiFiManager #959: autoconnect does not work after power failure or reset

gabi68, 2019-11-04:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x ] Development

**Esp8266/Esp32:**
- [ x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x ] 2.4.0
- [ ] staging (master/dev)

### Description

All is working as it should be, until board loose power or is reseted. Parameter input (postcode) in the wifi setup screen are not retained after power failure or reset. Please somebody have a look and tell what is wrong and what do I need to write to fix it.

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

#ifdef ESP32
#include <SPIFFS.h>
#endif

//define your default values here, if there are different values in config.json, they are overwritten.

char Postcode[15] = "";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setupSpiffs() {
  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
          Serial.println("\nparsed json");

          strcpy(Postcode, json["Postcode"]);

        } else {
          Serial.println("failed to load json config");
        }
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
 
}

void getwifi() {
  
  Serial.begin(115200);
  WiFi.enableInsecureWEP();
  Serial.println();

  setupSpiffs();

  // WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;

  //set config save notify callback
  wm.setSaveConfigCallback(saveConfigCallback);

  // setup custom parameters
  //
  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length

  WiFiManagerParameter custom_Postcode("Postcode", "Postcode", "", 15);

  wm.addParameter(&custom_Postcode);

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

//  wm.disconnect();
  if (!wm.autoConnect("YourClockAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(1000);
    // if we still have not connected restart and try all over again
    ESP.restart();
    delay(1000);
  }

  // always start configportal for a little while
  //wm.setConfigPortalTimeout(60);
  //WiFi.disconnect();
  wm.startConfigPortal("YourClockAP");

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  strcpy(Postcode, custom_Postcode.getValue());

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    
    json["Postcode"]   = Postcode;

   

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

    json.prettyPrintTo(Serial);
    json.printTo(configFile);
    configFile.close();

    shouldSaveConfig = false;
  }
  Serial.println();
  Serial.println(Postcode);
  postcode = String(Postcode);
 
}

### Debug Messages

```
20:49:30.453 -> mounting FS...
20:49:30.521 -> mounted file system
20:49:30.623 -> *WM: Adding parameter
20:49:30.623 -> *WM: Postcode
20:49:30.623 -> *WM: 
20:49:30.623 -> *WM: AutoConnect
20:49:30.623 -> *WM: Connecting as wifi client...
20:49:30.623 -> *WM: Using last saved values, should be faster
20:49:34.423 -> *WM: Connection result: 
20:49:34.423 -> *WM: 3
20:49:34.423 -> *WM: IP Address:
20:49:34.423 -> *WM: 192.168.1.2
20:49:34.490 -> *WM: SET AP STA
20:49:34.490 -> *WM: 
20:49:34.490 -> *WM: Configuring access point... 
20:49:34.490 -> *WM: YourClockAP
20:49:34.965 -> *WM: AP IP address: 
20:49:34.965 -> *WM: 192.168.4.1
20:49:34.999 -> *WM: HTTP server started
20:50:05.617 -> *WM: Request redirected to captive portal
20:50:05.651 -> *WM: Handle root
20:50:05.889 -> *WM: Request redirected to captive portal
20:50:05.889 -> *WM: Handle root
20:50:10.679 -> *WM: Handle root
20:50:15.939 -> *WM: Scan done
20:50:15.939 -> *WM: DIRECT-am[TV]UE46ES6800
20:50:15.939 -> *WM: -40
20:50:15.939 -> *WM: TALKTALK-1C1CA4
20:50:15.939 -> *WM: -53
20:50:15.939 -> *WM: TALKTALK167DDE
20:50:15.939 -> *WM: -62
20:50:15.939 -> *WM: BTHub5-QH8P
20:50:15.939 -> *WM: -91
20:50:15.939 -> *WM: Sent config page
20:50:34.477 -> *WM: WiFi save
20:50:34.477 -> *WM: Parameter
20:50:34.477 -> *WM: Postcode
20:50:34.477 -> *WM: rh106da
20:50:34.477 -> *WM: Sent wifi save page
20:50:36.477 -> *WM: Connecting to new AP
20:50:36.477 -> *WM: Connecting as wifi client...
20:50:36.511 -> *WM: Already connected. Bailing out.
20:50:36.544 -> Should save config
20:50:36.544 -> connected...yeey :)
20:50:36.544 -> saving config
20:50:36.714 -> {
20:50:36.714 ->   "Postcode": "rh106da"
20:50:36.714 -> }
20:50:36.714 -> rh106da

```

tablatronix, 2019-11-05:
Are you actually saving into spiffs and reading back out?

gabi68, 2019-11-05:
That is what I am trying to do. The "Postcode" value must be input by user and will be used in a weather api. In a case of power failure I need that value which is stored in config to be used for autoconnect. How should I do that?

gabi68, 2019-11-05:
Is there a way to save the credentials and parameters somewhere? 
When wifimanager is set to autoconnect to read that file (with all data necessary for connection).

tablatronix, 2019-11-05:
wifimanger does not itself store parameters, you have to do it yourself in your code. It only saves the wifi creds ( which is built into esp )

The spiffs example should already do that, and be a good starting point

tablatronix, 2019-11-05:
Your sketch should work , you are copying postcode out of the json file into `PostCode`

`strcpy(Postcode, json["Postcode"]);`

gabi68, 2019-11-05:
It's working as it should be, but when esp8266 is in reset or has a power outage everything is going south. The credentials are saved, but the parameter is not. How can I save that parameter and reconnect with (autoconnect) with previous parameter (postcode).

tablatronix, 2019-11-05:
It should be saved in spiffs, add a debug and see if you can read it, if not do a full erase and try it again

tarontop, 2021-07-09:
> 它应该保存在 spiffs 中，添加调试并查看是否可以读取它，如果不能完全擦除并重试

I tried non-blocking, `wm.setConfigportalTimeout()` does not seem to work