tzapu/WiFiManager #960: Can't reset the configs

cacs200281, 2019-11-05:
Hello dears,
I am a newbie and so sorry if I'm doing some mistakes but I'm very need your help cause I must deliver my project at University this week.

My problem is when I use this line at my code: wifiManager.resetSettings();
After restart using this line: ESP.restart() that ESP still conects on my router  :/

My preliminary infos are:
Arduino IDE using an ESP-07 configured as Generic esp8266
Board manager: esp8266 by comunity V2.5.2
Library manager: wifimanager by tzapu V 0.14.0

At serial monitor I can see this msg:
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.

I saw some issues telling about this problem but when I tried to apply that solutions my board still doesn't to reset to the initial config like AP.

Thanks in advance for "any light on my way"

Regards

rick-pit, 2019-11-11:
Hi, i use eraseConfig instead because i had the same problem, but im not sure if erase all content is good for you... hope this help

```
ESP.eraseConfig();
  delay(1000);
//  ESP.reset();
  ESP.restart();
```

cacs200281, 2019-11-11:
> Hi, i use eraseConfig instead because i had the same problem, but im not sure if erase all content is good for you... hope this help
> 
> ```
> ESP.eraseConfig();
>   delay(1000);
> //  ESP.reset();
>   ESP.restart();
> ```

Hey Rick, I will try your method and update here, thanks a lot!
I found a "stable" config using arduino ide 1.8.10, lib 0.12.0 and board 2.4.0
Regards

tablatronix, 2019-11-11:
Erase your flash and try again.

cacs200281, 2019-11-11:
> Erase your flash and try again.

Ok, now I am away from my home but will update my tests here a.s.a.p.
Thank you
Regards

tablatronix, 2019-11-12:
Usually if erase config fails, it is because memory is corrupt or changed since it was first written.

If it continues to not work let me know, I have not seen any other issues

cacs200281, 2019-11-13:
Hello dears,
The solutions showed by Rick-pit solved that problem, I kept the same initial config (ide arduino+lib version+board version).
Before to share here about my problem I tried to this "reset" attempt in my setup block:
 Using physical switch to reset the board, I don't write here but exists a "for" counting the time of that switch is pressed...5seconds and below is the code...
if(digitalRead(sw_reset) == 0)
    {
      Serial.println("");
      Serial.println("Resetando...");
      delay(10);
      WiFiManager wifiManager;
      wifiManager.resetSettings();
      SPIFFS.begin();
      SPIFFS.format();
      delay(2000);
      ESP.restart();                 
    }    

It forced to AP mode but the message "THIS MAY CAUSE AP NOT TO START....." still were showed by serial monitor.
Now my code is working well applying the solution by Rick-pit, thanks friend.

Thanks a lot dears, I will learning more about the amazings possibilities using ESP family + tzapu lib :dagger: 

Kind regards

Ruandv, 2020-01-14:
@tablatronix this is a NON issue and can be closed ;-)

closed 2020-01-15 by tablatronix