tzapu/WiFiManager #961: Lost custom parameters sometimes..

dontsovcmc, 2019-11-06:
### Basic Infos
I add 27 custom parameters and increase style tag... Some /wifi responses are good, sometimes return page without all parameters or return only a piece of them (!)

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [x] ESP01
- [x] ESP12F (nodemcu)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
PLATFORM: Espressif 8266 2.2.3 #cd6f5cc > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), framework-arduinoespressif8266 2.20502.0 (2.5.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0)

### Description

DEBUG_MAX: Mem looks good.

I test connections with 2-3 clients - no differences.
Open ConfigPortal with mobile phone - OK
Open ConfigPortal with PC CaptivePortal - OK
Open PC Chrome - Failed!
F5 - Failed!
F5 - OK

### Debug Messages

```
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [2] Scan is cached 10175 ms ago
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 14050 ms ago
*WM: [1] 11 networks found
*WM: [2] AP:  -47 Waterius_0.9.4      //WTF?!
*WM: [2] AP:  -56 do
*WM: [2] AP:  -56 doG
*WM: [2] AP:  -70 Smart_box-132
*WM: [2] AP:  -71 MGTS_GPON_8
*WM: [2] AP:  -74 mgts201
*WM: [2] AP:  -79 Beeline_2G_F2
*WM: [2] AP:  -83 MGTS_GPON_3
*WM: [2] AP:  -83 Onlime_205
*WM: [2] AP:  -91 Leva
*WM: [2] AP:  -91 ALARM
*WM: [3] Custom params 27
*WM: [3] Page len= 14233              //GOOD!
*WM: [3] Sent config page 
*WM: [3] -> captive.apple.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [2] Scan is cached 14300 ms ago
*WM: [2] Portal Timeout In 287 seconds

*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 4401 ms ago
*WM: [1] 11 networks found
*WM: [2] AP:  -44 Waterius_0.9.4        //WTF?
*WM: [2] AP:  -57 doG
*WM: [2] AP:  -58 do
*WM: [2] AP:  -68 MGTS_GPON_8
*WM: [2] AP:  -70 MGTS_GPON_5
*WM: [2] AP:  -75 Onlime_205
*WM: [2] AP:  -77 mgts201
*WM: [2] AP:  -78 Beeline_2G_F
*WM: [2] AP:  -83 MGTS_GPON_3
*WM: [2] AP:  -85 romper-stomper
*WM: [2] AP:  -86 Leva
*WM: [3] Custom params 27
*WM: [3] Page len= 14253                //GOOD!
*WM: [3] Sent config page 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [2] Scan is cached 4599 ms ago

*WM: [2] WiFi Scan ASYNC completed in 1645 ms
*WM: [2] WiFi Scan ASYNC found: 10
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 2115 ms ago
*WM: [1] 10 networks found
*WM: [2] AP:  -48 do
*WM: [2] AP:  -49 doG
*WM: [2] AP:  -64 mgts20
*WM: [2] AP:  -74 MGTS_GPON_5
*WM: [2] AP:  -77 Smart_box-132
*WM: [2] AP:  -86 Beeline_2G_F
*WM: [2] AP:  -86 Onlime_205
*WM: [2] AP:  -86 MGTS_GPON_8
*WM: [2] AP:  -91 Leva
*WM: [2] AP:  -91 MGTS_GPON_3
*WM: [3] Custom params 27
*WM: [3] Page len= 13751
*WM: [3] Sent config page 
*WM: [2] <- HTTP Root 
*WM: [3] -> bizujazvmr 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> nfbcojst 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> kzwvgmvgtaxs 
*WM: [2] <- Request redirected to captive portal 

*WM: [2] Scan is cached 14050 ms ago
*WM: [1] 11 networks found
*WM: [2] AP:  -47 Waterius_0.9.4      //WTF?!
*WM: [2] AP:  -56 do
*WM: [2] AP:  -56 doG
*WM: [2] AP:  -70 Smart_box-132
*WM: [2] AP:  -71 MGTS_GPON_8
*WM: [2] AP:  -74 mgts201
*WM: [2] AP:  -79 Beeline_2G_F
*WM: [2] AP:  -83 MGTS_GPON_3
*WM: [2] AP:  -83 Onlime_205
*WM: [2] AP:  -91 Leva
*WM: [2] AP:  -91 ALARM
*WM: [3] Custom params 27
*WM: [3] Page len= 14233              //GOOD
*WM: [3] Sent config page 

*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 45475 ms ago
*WM: [1] 10 networks found
*WM: [2] AP:  -48 do
*WM: [2] AP:  -49 doG
*WM: [2] AP:  -64 mgts201
*WM: [2] AP:  -74 MGTS_GPON_5
*WM: [2] AP:  -77 Smart_box-132
*WM: [2] AP:  -86 Beeline_2G_F
*WM: [2] AP:  -86 Onlime_205
*WM: [2] AP:  -86 MGTS_GPON_8
*WM: [2] AP:  -91 Leva
*WM: [2] AP:  -91 MGTS_GPON_3
*WM: [3] Custom params 27
*WM: [3] Page len= 9549               //WTF?!
*WM: [3] Sent config page 

Here is part of data without custom parameters: 
type='password' placeholder=''><hr><br/><br/><br/><button type='submit'

*WM: AP:  -87 MGTS_GPON_3520
MEM: 15856
*WM: Custom params 26
MEM: 15856
*WM: Page len= 9822   // BAD!

//remove JS code ~1500
*WM: Custom params 26
MEM: 15080
*WM: Page len= 12501  //GOOD!

MEM: 19056
*WM: Custom params 26
MEM: 15856
*WM: Page len= 14016  // Good!
```

//GOOD try: 
```
MEM: 19032
*WM: AP:  -86 Onlime_205
MEM: 18856
*WM: AP:  -89 Leva
MEM: 18296
*WM: Custom params 26
MEM: 15112
*WM: Page len= 12755
:wr 86 0
:wrc 86 86 0
:wr 14133 0
:wrc 986 14133 0
:ack 86
:ack 986
:wr 13147 986
:wrc 1072 13147 0
:ack 1072
:wr 12075 2058
:wrc 1072 12075 0
:ack 1072
:wr 11003 3130
:wrc 1072 11003 0
:ack 1072
:wr 9931 4202
:wrc 1072 9931 0
:ack 1072
:wr 8859 5274
:wrc 1072 8859 0
:ack 1072
:wr 7787 6346
:wrc 1072 7787 0
:ack 1072
:wr 6715 7418
:wrc 1072 6715 0
:ack 1072
:wr 5643 8490
:wrc 1072 5643 0
:ack 1072
:wr 4571 9562
:wrc 1072 4571 0
:ack 1072
:wr 3499 10634
:wrc 1072 3499 0
:ack 1072
:wr 2427 11706
:wrc 1072 2427 0
:ack 1072
:wr 1355 12778
:wrc 1072 1355 0
:ack 1072
:wr 283 13850
:wrc 283 283 0
MEM: 13288
*WM: Sent config page 
:ack 283
:rcl
:abort
:ur 1
WS:dis
:del
:ref 1
WS:av
:ref 2
:ur 2
:ref 2
:ur 2
:rn 353
:c0 1, 353
[HTTP] states route

```

tablatronix, 2019-11-06:
This sounds like another open issue that was running out of memory

dontsovcmc, 2019-11-06:
Also strange:

After ~5 connections/disconnections ESP lost all scanned networks.

There were 5. It shows 1 with empty name... 
F5 - return 5 networks..

```
MEM: 19904
*WM: WiFi Scan ASYNC started 
MEM: 30000
*WM: <- HTTP Wifi 
MEM: 21848
*WM: Scan is cached 25906 ms ago
MEM: 21896
*WM: 4 networks found
MEM: 21896
*WM: DUP AP: 
MEM: 21896
*WM: DUP AP: 
MEM: 21896
*WM: DUP AP: 
MEM: 21648
*WM: AP:  0 
MEM: 21072
```
```
*WM: <- HTTP Wifi 
MEM: 22104
*WM: Scan is cached 30678 ms ago
MEM: 22152
*WM: 3 networks found
MEM: 21904
*WM: AP:  -46 do
MEM: 21656
*WM: AP:  -48 doG
MEM: 21312
*WM: AP:  -71 MGTS_GPON_8
MEM: 20848
*WM: Custom params 26
```

dontsovcmc, 2019-11-06:
Additional tryes:
```
WM: custp len= 3064
MEM: 18024
*WM: Page len= 7924   //BAD 
MEM: 16584

*WM: AP:  -96 SPS-3
MEM: 16328
*WM: Custom params 27
MEM: 16040
*WM: custp len=
....
MEM: 16328
*WM: Page len= 8270     //BAD 
MEM: 14656

*WM: AP:  -90 Blackvue750S-8335D6
MEM: 17584
*WM: Custom params 27
MEM: 17296
*WM: custp len=
MEM: 17128
*WM: custp len= 163
MEM: 17048
*WM: custp len= 3064
MEM: 17064
*WM: Page len= 6200     //BAD 
MEM: 16144
*WM: Sent config page
MEM: 24400
*WM: -> www.msftconnecttest.com
MEM: 24520

MEM: 17280
*WM: AP:  -93 Blackvue750S-8335D6
MEM: 16800
*WM: Custom params 27
MEM: 16512
*WM: custp len=
MEM: 16344
*WM: custp len= 163
MEM: 16264
*WM: custp len= 3064     //BAD 
MEM: 15832
*WM: Page len= 7956
MEM: 15360
*WM: Sent config page
```

GOOD
```
*WM: AP:  -89 Blackvue750S-8335D6
MEM: 19040
*WM: Custom params 27
MEM: 18752
*WM: custp len=
......
*WM: custp len= 2986
MEM: 15720
*WM: custp len= 3064   
MEM: 15872
*WM: Page len= 11101  //GOOD
MEM: 15088

*WM: <- Request redirected to captive portal
MEM: 31168
*WM: <- HTTP Wifi
MEM: 21264
*WM: WiFi Scan completed in 2189 ms
MEM: 21264
*WM: 12 networks found
MEM: 21264
*WM: DUP AP: newsprin
MEM: 21080
*WM: AP:  -52 a2-tst
MEM: 20896
....
*WM: AP:  -91 MikroTik-BE86A6 CPI
MEM: 18728
*WM: Custom params 27
MEM: 18440
*WM: page len=
MEM: 18272
*WM: page len= 163
MEM: 18192
.........
*WM: page len= 3064
MEM: 15560
*WM: Page len= 11309       //GOOD
MEM: 15312
*WM: Sent config page

refresh 
*WM: page len= 3064
MEM: 18072
*WM: Page len= 11107     //GOOD
MEM: 17304
*WM: Sent config page
```

tablatronix, 2019-11-07:
This would be expected if stack is getting corrupted, I think the network scan struct gets created there

dontsovcmc, 2019-11-12:
@tablatronix here is a good explain of fragmentation: https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html
as you see "replace, concat... " functions are best fragmentation friends...
One of the possible way to decrease it: add __FlashStringHelper arguments for some functions.
Firstly, I should understand how can i measure fragmentation.

dontsovcmc, 2019-11-12:
I fork and run heap fragmentation test for ESP8266 https://github.com/dontsovcmc/cpp4arduino/tree/esp8266_platformio_heap

I think, we have WiFiManager heap test.

LOOK!

Available  | LargestBlock | Fragmentation
```
32560 32552 0.02
23704 19768 16.60
19784 14352 27.46
19176 12616 34.21
17208 8744 49.19
16968 8744 48.47
16968 8744 48.47
16200 6752 58.32
16040 4856 69.73
15112 4872 67.76
14112 4856 65.59
14080 3408 75.80
14280 4864 65.94
13848 2896 79.09
13576 2600 80.85
13336 5064 62.03
13344 4968 62.77
13344 4968 62.77
13344 4968 62.77
13344 4968 62.77
13184 3856 70.75
13168 2896 78.01
13152 3408 74.09
13136 1912 85.44
13136 1912 85.44
13136 1912 85.44
13136 1912 85.44
13136 1912 85.44
13136 1912 85.44
13040 1880 85.58
13040 1880 85.58  //look!
..... many times..
12536 1880 85.00
.....
```

tablatronix, 2019-11-12:
#518

tablatronix, 2019-11-12:
DEBUG_MAX already does memory logging fyi

dontsovcmc, 2019-11-13:
I was gonna do this output today, and you beat me to it. Thanks! I will report here.

dontsovcmc, 2019-11-13:
hm.. 

BAD: 
*WM: custp len= 3467
[MEM] free: 18304 | max: 14624 | frag:  18%
*WM: custp len= 3542
[MEM] free: 22184 | max: 14624 | frag:  27%
*WM: Page len= 7362
[MEM] free: 21416 | max: 14528 | frag:  26%
*WM: Sent config page
000:00:00:24:166  ERROR     (I2C) : end error:2

GOOD: 
[MEM] free: 18064 | max: 16432 | frag:   9%
*WM: custp len= 3467
[MEM] free: 17936 | max: 16352 | frag:   9%
*WM: custp len= 3542
[MEM] free: 18120 | max: 11456 | frag:  32%
*WM: Page len= 11423
[MEM] free: 16664 | max: 15456 | frag:   8%
*WM: Sent config page

think about refactoring code.
adding String page; page.reserve(12000); is not a good way

tablatronix, 2019-11-13:
I am pretty sure all of the memory issues are the page string variable.
So simply fixing that will help.

Ideas welcome, I never had frag more than 13%, how much custom params or html are you using ?

dontsovcmc, 2019-11-13:
@tablatronix about 28 custom parameters with long labels and custom JS =) You can see result page: https://waterius.ru/esp/config.html

dontsovcmc, 2019-11-14:
@tablatronix do you have custom callback example? 
I want to make pull request, but I added my pages into yours.. 
I need:
1. redefine handleRoot
2. clone handleWifi and handleWifiSave pages
Is it possible outside WiFiManager files?

Refactor #1:
    String        getParamOut();
    String        getScanItemOut();
    String        getStaticOut();

    change to:
    void          appendParamOut(String &);
    void          appendScanItemOut(String &);
    void          appendStaticOut(String &);

It's better to add page.reserve(PAGE_MAX_BYTES);

tablatronix, 2019-11-14:
```C++
  wm.setWebServerCallback(bindServerCallback);

void bindServerCallback(){
  wm.server->on("/custom",handleRoute);
  // wm.server->on("/info",handleRoute); // can override wm!
}

void handleRoute(){
  Serial.println("[HTTP] handle route");
  wm.server->send(200, "text/plain", "hello from user code");
}
```

dontsovcmc, 2019-11-23:
I think problem solved! Memory issuu. Thanks for debug.

closed 2019-11-23 by dontsovcmc