tzapu/WiFiManager #964: Cannot Connect to Config Portal when AP Config is Saved

lbussy, 2019-11-13:
There's been a number of issues posted which circle around this, I posted one and asked for it to be closed since frankly, I was having a hard time reproducing.  I can give you the workflow that I think demonstrates the issue, however, I am remote from my dev setup for the rest of the week so actually working with the portal will have to wait.

Here is the workflow which demonstrates the issue:

1. Properly configure the controller to connect to a given AP via captive portal
1. Verify connectivity wifiManager.autoConnect
1. Disable AP and restart controller
1. Observe controller timeout on AP connection, and start config portal
1. Attempt to connect to AP and receive failure (different messages depending upon platform)

This *seems* to not be new information, judging by the below section of code (specifically @ 541):

https://github.com/tzapu/WiFiManager/blob/2447a3fd7408c5718997b3fce75cdbcca2b4ce0e/WiFiManager.cpp#L532-L543

It's possible that I'm "doing it wrong" but I believe my code is very aligned to your examples.  The intent of this method is to provide a fall-back if the AP cannot be connected to, but that fails if the AP was configured previously so that seems at odds with your intent.

Is there something else I can provide to show you what I mean?

tablatronix, 2019-11-13:
No idea, without any information whatsovever, it is hard to know what is going on.

lbussy, 2019-11-13:
That actually is a lot of information.  You have a note there in your own code that says:

```
// @todo even if sta is connected, it is possible that softap connections will fail, IOS says "invalid password", windows says "cannot connect to this network" researching
```

Are you researching?  Because that's exactly what's happening.

My steps to reproduce are environmental, not code-based.  What information is it you think would be more prescriptive?

tablatronix, 2019-11-13:
The entire form you deleted when you opened this issue for starters. versions, lib, serial logs..

tablatronix, 2019-11-13:
also you say AP 4 times, and I have no idea which ap you are talking about , some are the router ssid, some the wm portal.. again , i have no idea what issue you are having.

I Dont even know if you are using esp8266 or esp32...

lbussy, 2019-11-13:
I didn't delete the form, I was surprised I was not presented with one.  I'll leave debugging why that didn't happen to the GitHub people.

I also apologize - when I said I opened an issue here previously, I was mistaken.  We discussed this in #866 and I left to try out the development branch.

Here is the typical information I used in a diffferent repo which should satisfty the immediate need:

#### Platform

- Hardware:			ESP-12S
- Core Version:			SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
- Development Env:		PlatformIO
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			qio
- Flash Size: 		4Mb
- lwip Variant:   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   460800

As I said I am using your development branch.

Allow me to restate the steps to reproduce using your terms:

1. Properly configure the ESP8266 (WemosD1) controller to connect to a given router SSID via captive portal
1. Verify connectivity (to access point) via wifiManager.autoConnect
1. Disable router and restart the controller
1. Observe controller timeout on router SSID connection, and start your captive portal
1. Attempt to connect to the captive portal and receive failure (different messages depending upon platform)

Does that help?

tablatronix, 2019-11-13:
No form ? that is indeed interesting. If you click new issue you are supposed to have prefilled info form..  Sorry about that.

Can you elaborate on the messages you are seeing?
Can you try esp staging as there was some fatal ap things fixed.
Also when in doubt I do a flash erase to clean the esp out.

tablatronix, 2019-11-13:
[image]

lbussy, 2019-11-13:
I clicked on the new issue button again when you said that and it was there, it just didn’t pop up when I opened the issue.  Strange, but that’s why I said it was a GitHub thing.   That or I just happened to hit an amazingly accurate series of keystrokes as the page was loading.

When I get home on Friday I can try staging as well.  It’s tough to walk the wife at home through connecting to a captive portal no matter how many times we stay in a hotel. 

I’ll also post some screen grabs when it fails to connect and turn up the logging so it captures something interesting. 

There’s a 98% chance I tried a flash erase since that’s what I always tell my users but I’ll do that again as well.

lbussy, 2019-11-16:
You seem to have pushed some new changes.  I've got a new WAP here to test with however the portal now fails to display for me (#970).  So, not sure if I can reproduce this till that's resolved.

tablatronix, 2019-11-17:
The portal does not show via what ip address? How are you testing

lbussy, 2019-11-17:
This can be closed (I'll go ahead and do that.)  Since updating to the latest from the development branch I cannot reproduce the issue I was having.

closed 2019-11-17 by lbussy