tzapu/WiFiManager #967: configPortalHasTimeout dependings

dontsovcmc, 2019-11-15:
I can't understand, why we need to call handleRequest() in root handle to reset _webPortalAccessed?
configPortal timeout start point should be the time when it starts. But now we have another timepoint - when portal opened by client. 
Why does it needed?!

If we make timeout only from function call, we simplify configPortalHasTimeout() function and remove handleRequest().

tablatronix, 2019-11-15:
That is how I hacked the webactive timeout, it was easier than adding new variables for something that is experimental and not even enabled by default. It is a bit of a kludge, it also prevents me from having to modify the timeout evaluation, easier to just bump start time, we do not actually need it..

tablatronix, 2019-11-15:
https://github.com/tzapu/WiFiManager/blob/development/WiFiManager.h#L246