tzapu/WiFiManager #968: dublicate function: /0wifi and /wifi&refresh

dontsovcmc, 2019-11-15:
If better to remove dublicate functions to decrease code size. I think /0wifi is excess, cause we can change button code:
`"<form action='/wifi?refresh=1'    method='get'><button>Configure WiFi</button></form><br/>\n", // MENU_WIFI`

no scan will be /wifi without argument:
`"<form action='/wifi'   method='get'><button>Configure WiFi (No Scan)</button></form><br/>\n", // MENU_WIFINOSCAN`

tablatronix, 2019-11-15:
it is legacy

closed 2019-11-15 by tablatronix

dontsovcmc, 2019-11-15:
WiFiManager has bad architecture, any customization are hard to do. I want to see anybody who use this methods outside WiFiManager source code. It's difficult to refactor library if we have a lot of unused features.

tablatronix, 2019-11-15:
You keep saying "refactor" refator what?
What needs refactoring, I just refactored the entire thing to start organizing for future updates, v0 has no abstraction at all.

dontsovcmc, 2019-11-15:
I mean less code, less errors only =)