tzapu/WiFiManager #969: Continues Integration: automatic hardware tests

dontsovcmc, 2019-11-15:
It's comfortable way to refactor library is to have a lot of tests.
I can write small software tests, but hardware tests are more interesting.

Idea: 
1. use platformio for command line build WiFiManager test examples
2. connect to ESP8266 using python or bash language. For example:
https://github.com/rockymeza/wifi
3. test web pages content by HTTP requests
4. check items, push buttons, fill labels by SeleniumHQ and etc

Pros: 
1. We can do a lot of refactor safety.

Cons: 
1. Control PC Wi-Fi is platform dependency.

dontsovcmc, 2019-12-02:
@tablatronix I made it =)
Now we can check pages and refactor code safety:
https://github.com/dontsovcmc/ESPWebServerTest

What is you OS? Now I support MacOS Wi-fi adapter.

dontsovcmc, 2019-12-05:
Look at` examples/DEV/OnDemandConfigPortal` test:
https://github.com/dontsovcmc/ESPWebServerTest#wifimanager-example-test

here is the Test:
https://github.com/dontsovcmc/ESPWebServerTest/blob/master/examples/WifiManagerTest/OnDemandConfigPortal/test_configportal.py

note: please, move handleRoute(), getTime() functions upper, cause compile error

tablatronix, 2019-12-05:
What compile error?

dontsovcmc, 2019-12-05:
In C++ you should define function before use (PlatformIO works as C++). 
Arduino IDE hacks this rule. So, add 
```
void handleRoute();
void getTime(); 
```
before functions or mode them body upper.

tablatronix, 2019-12-05:
where?

dontsovcmc, 2019-12-05:
https://github.com/tzapu/WiFiManager/blob/development/examples/DEV/OnDemandConfigPortal/OnDemandConfigPortal.ino

tablatronix, 2019-12-05:
I use platformio, I get no compile errors.

tablatronix, 2019-12-05:
Are you compiling a copy or stream as cpp not ino ?

dontsovcmc, 2019-12-05:
Rename ino to cpp and compile... Platformio can eat ino?!

tablatronix, 2019-12-05:
Yes

dontsovcmc, 2019-12-05:
Nice, simplify script.