tzapu/WiFiManager #970: Configuration Items in Portal Fail to Display

lbussy, 2019-11-16:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development @ 22e35b2

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X] ESP12 E/F/S (Wemos D1 mini v2.1.0)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [X] 2.5.2=20502000
- [ ] staging (master/dev)

### Description

I am trying to reproduce the issues I have been having in #964 and have run into what I think is a newly introduced issue.  At least I hope it is or lese I have lost my mind.

The captive portal fails to display AP configuration fields.  I see the following only after connecting to the controller's AP:

[image]

### Settings in IDE

Module: Wemos D1 Mini

Additional libraries: NONE

### Sketch

```cpp
#include <WiFiManager.h>

void setup() {
    delay(1000);
    Serial.begin(74880);
    Serial.setDebugOutput(true);
    Serial.flush();

    WiFi.mode(WIFI_STA);
    delay(200);

    WiFiManager wm;
    if (!wm.autoConnect()) {
        Serial.println(F("Failed to connect and hit timeout."));
        ESP.restart();
        delay(1000);
    }
    Serial.println(F("Connected."));
}

void loop() {}
```

### Debug Messages

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] No saved credentials, skipping wifi
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED
*WM: [2] Starting Config Portal
*WM: [3] WIFI station disconnect
*WM: [3] WiFi station enable
del if0
usl
mode : null
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  ESP_245A77
modwifi evt: 8
e : softAP(5e:cf:7f:24:5a:77)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
wifi evt: 8
*WM: [2] AP has anonymous access!
[AP] softap config unchanged
*WM: [1] SoftAP Configuration
*WM: [1] --------------------
*WM: [1] ssid:             ESP_245A77
*WM: [1] password:
*WM: [1] ssid_len:         10
*WM: [1] channel:          1
*WM: [1] authmode:
*WM: [1] ssid_hidden:
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] --------------------
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started
mode : sta(5c:cf:7f:24:5a:77) + softAP(5e:cf:7f:24:5a:77)
add if0
wifi evt: 8
scandone
*WM: [2] WiFi Scan completed in 2198 ms
*WM: [2] Config Portal Running, blocking, waiting for clients...
wifi evt: 7
add 1
aid 1
station: 8c:86:1e:39:36:40 join, AID = 1
wifi evt: 5
wifi evt: 7
wifi evt: 7
wifi evt: 9
wifi evt: 7
wifi evt: 7
*WM: [3] -> captive.apple.com
*WM: [2] <- Request redirected to captive portal
*WM: [2] <- HTTP Root
*WM: [3] -> 192.168.4.1
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 5907 ms ago
*WM: [3] -> captive.apple.com
*WM: [2] <- Request redirected to captive portal
*WM: [2] <- HTTP Root
*WM: [3] -> 192.168.4.1
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 6353 ms ago
*WM: [3] -> captive.apple.com
*WM: [2] <- Request redirected to captive portal
*WM: [2] <- HTTP Root
*WM: [3] -> 192.168.4.1
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [2] Scan is cached 6451 ms ago
wifi evt: 9
wifi evt: 7
wifi evt: 7
wifi evt: 7
```

tablatronix, 2019-11-17:
I have never seen this before..
So just no menu at all? 

Is it reproducible? Can you bump debugging level?

tablatronix, 2019-11-17:
Can you try a different device? I wonder if its an html rendering issue

lbussy, 2019-11-17:
No menu at all, just what you see in that screenshot.  I tried it with two different phones and neither one has taken a recent update.  I also opened a new board and tried it and had the same results.  I turned on `DEBUG_ESP_CORE` and `DEBUG_ESP_HTTP_SERVER` (in addition to `DEBUG_ESP_WIFI`, I hope that's what you meant.  

Here's the debug:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
data  : 0x3ffe8000 ~ 0x3ffe850c, len: 1292
rodata: 0x3ffe8510 ~ 0x3ffe920c, len: 3324
bss   : 0x3ffe9210 ~ 0x3ffef418, len: 25096
heap  : 0x3ffef418 ~ 0x3fffc000, len: 52200
*WM: [1] getCoreVersion():          2_5_2
*WM: [1] system_get_sdk_version():  2.2.1(cfd48f3)
*WM: [1] system_get_boot_version(): 31
*WM: [1] getFreeHeap():             50256
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] No saved credentials, skipping wifi
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED
*WM: [2] Starting Config Portal
*WM: [3] WIFI station disconnect
*WM: [3] WiFi station enable
del if0
usl
mode : null
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  ESP_773D7D
modwifi evt: 8
e : softAP(2e:f4:32:77:3d:7d)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
wifi evt: 8
*WM: [2] AP has anonymous access!
[AP] softap config unchanged
*WM: [1] SoftAP Configuration
*WM: [1] --------------------
*WM: [1] ssid:             ESP_773D7D
*WM: [1] password:
*WM: [1] ssid_len:         10
*WM: [1] channel:          1
*WM: [1] authmode:
*WM: [1] ssid_hidden:
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] --------------------
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started
mode : sta(2c:f4:32:77:3d:7d) + softAP(2e:f4:32:77:3d:7d)
add if0
wifi evt: 8
wifi evt: 7
[... 296 lines cut ...]
:wr 3281 0
:wrc 965 3281 0
:wr 2316 965
:ack 107
:ack 965
:wr 2316 965
:wrc 1072 2316 0
:ack 1072
:wr 1244 2037
:wrc 1072 1244 0
:ack 1072
:wr 172 3109
:wrc 172 172 0
*WM: [2] Scan is cached 8636 ms ago
:ack 172
:rcl
:abort
:ur 1
WS:dis
:del
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
```

tihoangyeudau, 2019-11-17:
i have same issue.

tablatronix, 2019-11-17:
ok let me see if I broke something

lbussy, 2019-11-17:
Confirmed fixed by updating from 22e35b2 to 56ad40d.

closed 2019-11-17 by lbussy

tablatronix, 2019-11-17:
Sorry about that

tihoangyeudau, 2019-11-19:
thank you all friend

commit 56ad40d0 (2019-11-17): Fixes #970 I broke constructor