tzapu/WiFiManager #971: can't compile with esp8266 v2.6.1

hznupeter, 2019-11-17:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
- [x]2.6.1
### Description

Problem description
can't compile with esp8266 2.6.0 and 2.6.1

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch
Example:AutoConnect
```cpp
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}
void loop() {
    // put your main code here, to run repeatedly:
}
```

### Debug Messages

```
In file included from D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,
                 from D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/Arduino.h:32,
                 from sketch\AutoConnect.ino.cpp:1:
D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))                                                                                                                    
D:\GitHub\Mixly1.0_WIN\arduino\portable\sketchbook\libraries\WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM'
 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta charset=\"UTF-8\" name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";
                        ^
In file included from D:\GitHub\Mixly1.0_WIN\arduino\portable\sketchbook\libraries\WiFiManager\examples\AutoConnect\AutoConnect.ino:5:0:
D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
"WiFiManager.h" 对应多个库
 已使用： D:\GitHub\Mixly1.0_WIN\arduino\portable\sketchbook\libraries\WiFiManager
"ESP8266WiFi.h" 对应多个库
 已使用： D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WiFi
"DNSServer.h" 对应多个库
 已使用： D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\DNSServer
"ESP8266WebServer.h" 对应多个库
 已使用： D:\GitHub\Mixly1.0_WIN\arduino\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer
exit status 1
为开发板 WeMos D1 R1 编译时出错。

```

tihoangyeudau, 2019-11-17:
WIFIMANAGER_DEV STILL WORK WELL WITH 2.6.1. YOU SHOULD TRY OTHER CODE

tablatronix, 2019-11-17:
https://github.com/tzapu/WiFiManager/releases/tag/0.15.0-beta

closed 2019-11-17 by tablatronix