tzapu/WiFiManager #972: Error compiling for board WeMos D1 R1

belazelkos, 2019-11-28:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] 2.6.2
- [ ] staging (master/dev)

### Description

I have problem to compile WiFiManager for Wemos D1 mini board, I even try with Example AutoConnect with out any luck (it's the same even if I select some other board for example Generic ESP8266 board).
All of the boards definitions are updated and all of the libraries are updated to the latest stable releases.

Any help please!

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}
```

### Debug Messages

```
In file included from C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,

                 from C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

                 from C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/Arduino.h:32,

                 from sketch\AutoConnect.ino.cpp:1:

C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))

                                                                                                                                 ^

C:\Users\B\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM'

 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                        ^

In file included from C:\Users\B\Documents\Arduino\libraries\WiFiManager\examples\AutoConnect\AutoConnect.ino:5:0:

C:\Users\B\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

exit status 1
Error compiling for board WeMos D1 R1.
```

PaoloP74, 2019-11-28:
Open the library .c and .h file and replace all occurrences of HTTP_HEAD with HTTP_MYHEAD.
HTTP_HEAD has become a reserved word.

tablatronix, 2019-11-28:
Its fixed in the beta...

closed 2019-11-28 by tablatronix