tzapu/WiFiManager #973: Scanning for hidden SSID

trlafleur, 2019-11-29:
We recently did a pull request suggesting a change to WiFiMulti to allow scanning for hidden SSID...

We have a version called WiFiMulti2 that we have used in a number of projects that incorporate these changes...

You might find this of interest for your work here...

~~~
https://github.com/espressif/arduino-esp32/pull/3464
~~~

tablatronix, 2020-01-23:
#956

tablatronix, 2020-01-23:
hmm, see my comments on the pr, not sure I understand this

trlafleur, 2020-01-23:
The standard WiFiMulti code will scan for all SSID it can see, and connected you to one that you have requested if it is available... It will show you some information about the hidden SSID, BUT! if the SSID you requesting is hidden, it will NOT connect you to that SSID. The logic in the code is wrong.

The modification we made to WiFiMulti in the pull request above, will connect you to a hidden SSID if it is available....

tablatronix, 2020-01-23:
You say the logic is wrong, but all the PR says is that you add a delay ( among other connection toggling hacks)