tzapu/WiFiManager #979: ESP32 shows captive portal every second time

ElToberino, 2019-12-08:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x ] Other

**ESP32 Core Version:
current

### Description

There is a problem with WiFi-Connections of ESP32 in combination with certain routers/APs.
The issue is described here: [https://github.com/espressif/arduino-esp32/issues/2501#issue-413808401](https://github.com/espressif/arduino-esp32/issues/2501#issue-413808401)
A WiFi connection can only be established on every second attempt. A far as I know, the reason for that bug is still unknown. There are some workarounds for that problem, but they are not perfect and they don't work using WiFiManager.
WiFiManager (with saved credentials!) connects only every second time - the other times it starts the ESP32-AP with the captive portal.
If you're facing this ESP32 issue, you can't use WifiManager. I know that this issue isn't caused by WifiManager itself, but I'd like to ask if a workaround for that issue could be implemented or how I could use WifiManager anyway.

tablatronix, 2019-12-08:
This sounds like a not real bug, or a very specific edge case. 

Have you done an NVS erase?
Are you using the lastest staging esp32 lib ?
Are you using rev 0 hardware?

tablatronix, 2019-12-08:
You would have to provide some detailed logs with debugging on and maybe wireshark logs if possible.

TheNitek, 2020-02-16:
I'm having the same issue using my ESP32 with a FritzBox. Without the WifiManager I just have to call WiFi.disconnect(); once if the connection attempt fails and try to connect a second time. With WifiManager (develeopment branch) I'll just end up in AP mode.

Log looks like this:
*WM: [2] ESP32 event handler enabled
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to SAVED AP: ***
*WM: [3] Using Password: ***
*WM: [3] WiFi station enable 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] [EVENT] WIFI_REASON: 202
*WM: [2] Connection result: WL_CONNECT_FAILED
*WM: [3] lastconxresulttmp: WL_STATION_WRONG_PASSWORD
*WM: [3] lastconxresult: WL_STATION_WRONG_PASSWORD
*WM: [1] AutoConnect: FAILED
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID

TheNitek, 2020-02-16:
This seems to work:

```
  wifiManager.setEnableConfigPortal(false);
  if(!wifiManager.autoConnect()) {
    Serial.println("Retry autoConnect");
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    wifiManager.setEnableConfigPortal(true);
    wifiManager.autoConnect("NaeveBot", "botbotbot");
  }

```
(For some reason I had to add the ```WiFi.mode(WIFI_OFF);``` line to make it work)

tablatronix, 2020-02-16:
So it sounds like autoconnect is not connecting for some reason the first time, it says connect failure 302, are you using an up to date esp32 lib? 

This is rather strange and almost defiantly a esp32 bug or race condition. 

Did you erase nvs ?

TheNitek, 2020-02-18:
I am up to date on wifimanager and esp32 platform.

I'm quite sure it's a platform issue ( https://github.com/espressif/arduino-esp32/issues/2501#issue-413808401 ) 

I don't think the configportal timeout will help, because it still won't be connected

tablatronix, 2020-02-18:
I never have this issue, are you sure you guys are erasing your esps

tablatronix, 2020-02-18:
try now, fetch latest

TheNitek, 2020-02-19:
Works like a charm now using setCleanConnect(true). Thank you very much!

tablatronix, 2020-02-19:
That is what it was before, I turned it off... lol you just turned it back on

TheNitek, 2020-02-19:
Ok, now you got me confused :-D 
So I gave it another try and actually neither "true" nor "false" work: After every 2 seconds reset the portal opens up. If I just reset again (without doing anything) it connects fine.

Still wonder what I tested earlier but probably it(/I) didn't flash the most current firmware

tablatronix, 2020-02-19:
Interesting, I was having issues with reconnects also but they went away when I removed that disconnect. Could be a race condition, what if you add a delay(2000) before autoconnect?

tablatronix, 2020-02-19:
Do you have anything in your code that could be starting wifi before wm starts wifi, sometimes that causes issues.

You might also want to enable esp wifi debugging and log everything

TheNitek, 2020-02-20:
There is nothing happening before the wm starts and a delay does not seem to help. Me feeling is that his is highly depending on the access point type and it's key handling.
I'll try to collect debugging logs.

TheNitek, 2020-02-20:
```
  wifiManager.setEnableConfigPortal(false);
  if(!wifiManager.autoConnect()) {
    Serial.println("Retry autoConnect");
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    wifiManager.setEnableConfigPortal(true);
    wifiManager.autoConnect();
  }
```

results in

```
Connecting Wifi
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
Retry autoConnect
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
```

Wild guess: The ESP stores some credentials for faster reconnection without full authentication with the WiFi after DeepSleep/Reset. These seem to expire for some reason (maybe that's where the router model comes into play) and the AUTH_FAIL is required for the ESP to realize that is has to connect using the PSK again.

tablatronix, 2020-02-20:
Never heard of anything like that, are you using deep sleep? or just reseting, can you post a full log of it working then not working ? do you have esp debugging turned on ?

TheNitek, 2020-02-20:
I read about this quite some time ago, so I am not 100% sure I remember correctly.
```
Connecting Wifi
I (41) wifi: wifi driver task: 3ffb5fe4, prio:23, stack:3584, core=0
I (328) wifi: wifi firmware version: 7997e4b
I (329) wifi: config NVS flash: enabled
I (329) wifi: config nano formating: disabled
I (329) wifi: Init dynamic tx buffer num: 32
I (333) wifi: Init data frame dynamic rx buffer num: 32
I (338) wifi: Init management frame dynamic rx buffer num: 32
I (344) wifi: Init management short buffer num: 32
I (348) wifi: Init static rx buffer size: 1600
I (353) wifi: Init static rx buffer num: 16
I (356) wifi: Init dynamic rx buffer num: 32
I (428) wifi: mode : sta (3c:71:bf:6b:d5:90)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
I (620) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1357) wifi: state: init -> auth (b0)
I (1362) wifi: state: auth -> init (8a0)
I (1363) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
Retry autoConnect
I (1400) wifi: flush txq
I (1401) wifi: stop sw txq
I (1403) wifi: lmac stop hw txq
I (1413) wifi: mode : sta (3c:71:bf:6b:d5:90)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
I (1545) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1546) wifi: state: init -> auth (b0)
I (1551) wifi: state: auth -> assoc (0)
I (1558) wifi: state: assoc -> run (10)
I (1589) wifi: connected with handtuch, channel 1, BW20
I (1594) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.180, MASK: 255.255.255.0, GW: 192.168.1.1
WiFi connected with IP: 192.168.1.180
```

Currently I am not using deep sleep, just reset. The code I posted is the only wifi related code.

tablatronix, 2020-02-20:
Nice, and you are using esp32 staging from github?

tablatronix, 2020-02-20:
Have you tried disabling sleep.

TheNitek, 2020-02-20:
It's release 1.11.2 through platform.io

What do you mean by "disabling sleep"?

tablatronix, 2020-02-20:
`    WiFi.setSleep(false);`

tablatronix, 2020-02-20:
You did a full erase etc?

TheNitek, 2020-02-20:
Disabling sleep seems to do the trick:

```
Connecting Wifi
[W][WiFiGeneric.cpp:566] setSleep(): STA has not been started
I (42) wifi: wifi driver task: 3ffb5fe4, prio:23, stack:3584, core=0
I (330) wifi: wifi firmware version: 7997e4b
I (331) wifi: config NVS flash: enabled
I (331) wifi: config nano formating: disabled
I (331) wifi: Init dynamic tx buffer num: 32
I (335) wifi: Init data frame dynamic rx buffer num: 32
I (340) wifi: Init management frame dynamic rx buffer num: 32
I (346) wifi: Init management short buffer num: 32
I (350) wifi: Init static rx buffer size: 1600
I (355) wifi: Init static rx buffer num: 16
I (358) wifi: Init dynamic rx buffer num: 32
I (430) wifi: mode : sta (3c:71:bf:6b:d5:90)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
I (557) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (558) wifi: state: init -> auth (b0)
I (565) wifi: state: auth -> assoc (0)
I (573) wifi: state: assoc -> run (10)
I (610) wifi: connected with handtuch, channel 1, BW20
I (667) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.180, MASK: 255.255.255.0, GW: 192.168.1.1
WiFi connected with IP: 192.168.1.180
```

I don't do a full erase everytime I flash, no.

tablatronix, 2020-02-20:
no not everytime just once is good when changing sdk and esp versions

tablatronix, 2020-02-21:
https://github.com/espressif/arduino-esp32/issues/2501

Definitely something going on here in SDK. Seems to be something with NVS

tablatronix, 2020-02-21:
Let me know how stable you are now with no sleep

TheNitek, 2020-02-21:
Yesterday the no sleep looked fine, today it does not anymore (?!). I feel like it changed the behaviour from "every second time" to "every now and then"

tablatronix, 2020-02-22:
Lol to me that sounds like a race condition for sure. Very hard to debug, ill keep looking for hints, I wonder if we can turn off nvs and connect with credentials each time just to see

tablatronix, 2020-05-11:
I have been having issues where waitforconnectresult takes the entire timeout period, something is definitely wrong with routers allowing reconnect or esp lib..

tablatronix, 2020-05-25:
#1067

fhpa-git, 2020-09-15:
Hi to everybody,

I have/had the same issue too since many months. I am using a Fritzbox 7590. Up to now i always connected my ESP32 wifi through the fritzbox.

But just minutes ago i connected via an extra access point from Fritz-LAN to a different WLAN. And my problem went away. I checked an started many times...

The extra access point is a TP-LINK TL MR3020 V1.9

best regards & good luck
fhpa-git

tablatronix, 2020-11-19:
I added capability to add retries to autoconnect, hope it helps someone, feel free to test it

Patronics, 2021-05-03:
> I added capability to add retries to autoconnect, hope it helps someone, feel free to test it

This fixed (or successfully worked around, at least) this issue for me, thanks!

Interestingly, it seems to consistently fail twice, and succeed on the 3rd reconnect attempt. Also, when will you be releasing the next release? It adds some complexity to need to install from git instead of from the boards manager directly (and your 2.0.3 alpha release didn't have the retries feature).

closed 2022-01-11 by tablatronix

lyricnz, 2024-11-24:
FWIW I have the same symptom (connects every second boot, with 100% reliability).  Adding the retries didn't help for me. 

```
Connecting to WiFi
*wm:[2] AutoConnect 
*wm:[4] No Hostname to set 
*wm:[3] ESP32 event handler enabled 
*wm:[3] Connecting as wifi client... 
*wm:[4] STA static IP:
*wm:[3] setSTAConfig static ip not set, skipping 
*wm:[2] Connect Wifi, ATTEMPT # 1 of 3
*wm:[2] Connecting to SAVED AP: iot
*wm:[4] Using Password: REMOVED
*wm:[4] WiFi_enableSTA enable
*wm:[4] Mode after delay:  STA
*wm:[2] connectTimeout not set, ESP waitForConnectResult... 
*wm:[3] [EVENT] WIFI_REASON:  202
*wm:[3] [EVENT] WIFI_REASON:  202
*wm:[3] Connection result: WL_CONNECT_FAILED
*wm:[2] Connect Wifi, ATTEMPT # 2 of 3
*wm:[2] Connecting to SAVED AP: iot
*wm:[4] Using Password: REMOVED
*wm:[4] WiFi_enableSTA enable
*wm:[4] Mode after delay:  STA
*wm:[2] connectTimeout not set, ESP waitForConnectResult... 
*wm:[3] Connection result: WL_CONNECT_FAILED
*wm:[2] Connect Wifi, ATTEMPT # 3 of 3
*wm:[2] Connecting to SAVED AP: iot
*wm:[4] Using Password: REMOVED
*wm:[4] WiFi_enableSTA enable
*wm:[4] Mode after delay:  STA
E (8862) wifi:sta is connecting, return error
[  4270][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
*wm:[2] connectTimeout not set, ESP waitForConnectResult... 
*wm:[3] Connection result: WL_CONNECT_FAILED
*wm:[4] lastconxresulttmp: WL_STATION_WRONG_PASSWORD
*wm:[4] lastconxresult: WL_STATION_WRONG_PASSWORD
*wm:[2] AutoConnect: FAILED for  3654 ms
*wm:[3] Starting Config Portal 
*wm:[4] WiFi station disconnect 
*wm:[4] WiFi_enableSTA enable
*wm:[3] Disabling STA 
*wm:[3] Enabling AP 
*wm:[2] StartAP with SSID:  Info-Orbs_88
```