tzapu/WiFiManager #980: Break; Wifimanger

jelmo2, 2019-12-09:
Master
ESP8266
ESP-12
ESP Core Version:	2.4.0
 
Hi tzapu,

maybe this has already been in the forum but I couldn't find it. I am working on a project with the option to read data via the internet. now my question is that i want the entire wifi manager (wifi) to be switched off after a certain time if it is not cofigured by the user. the only thing I can find is a timeout for the config portal.

in brief:

I get stuck with the code if the WiFi is not set, here I want to give the user 5 minutes to configure the WiFi. after 5 minutes the device must simply execute its code without wifi.

maybe you can help me.

Code:
void setup(void) {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  Wire.begin(5,4); // change hardware I2C pins to (5,4) (D1,D2)
  int chk = DHT.read(dht);
  pinMode(RELAY, OUTPUT);
  digitalWrite(RELAY, 1);// Turns Relay On
  digitalWrite(LED_BUILTIN, LOW);
  u8g2.begin(); // init display
//-------------------------------------------  
//Versie firmware 
 u8g2.clearBuffer();         // clear the internal memory  
 u8g2.setFont(u8g2_font_helvB08_tr);
 u8g2.drawStr( 0, 10, "JDJ electronics");
 u8g2.drawStr( 0, 30, "Model: DM");
 u8g2.drawStr( 0, 40, "Firmware: v2.5"); 
 u8g2.drawStr( 0, 50, "251019"); //firmware date  
 u8g2.drawStr( 10, 60, "2019");
 u8g2.setCursor(0, 60);
 u8g2.setFont(u8g2_font_helvB08_tf);
 u8g2.print(char(169)); // copyright karakter
 u8g2.sendBuffer();          // transfer internal memory to the display 
 delay(3000);
   
//Wifi setup + oled
 u8g2.clearBuffer();         // clear the internal memory  
 u8g2.setCursor(15, 36);
 u8g2.setFont(u8g2_font_open_iconic_www_2x_t);
 u8g2.print(char(81)); // WIFI icon
 u8g2.setFont(u8g2_font_helvB08_tr);
 u8g2.drawStr( 40, 32, "Connect to");
 u8g2.drawStr( 40, 45, "JDJ-DM");
 u8g2.sendBuffer();          // transfer internal memory to the display  
 
  WiFiManager wifiManager;
  wifiManager.autoConnect("JDJ-DM");
  wifiManager.setConfigPortalTimeout(60);
    if(!wifiManager.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();   
  }
  
  Serial.println("Connected.");
  digitalWrite(LED_BUILTIN, HIGH);
  server.begin();

 server.on("/", handle_OnConnect);
 server.onNotFound(handle_NotFound);

 server.begin();
 Serial.println("HTTP server started");
 Serial.println(WiFi.localIP());
 
//Connected Succes 
 u8g2.clearBuffer();         // clear the internal memory  
 u8g2.setCursor(15, 36);
 u8g2.setFont(u8g2_font_open_iconic_www_2x_t);
 u8g2.print(char(81)); // WIFI icon
 u8g2.setFont(u8g2_font_helvB08_tr);
 u8g2.drawStr( 40, 32, "Connected");
 u8g2.setCursor( 40, 45);
 u8g2.print(WiFi.localIP());
 u8g2.sendBuffer();          // transfer internal memory to the display 
 delay(1000); 
 
//Check up
 u8g2.clearBuffer();         // clear the internal memory  
 u8g2.setCursor(15, 36);
 u8g2.setFont(u8g2_font_open_iconic_embedded_2x_t);
 u8g2.print(char(66)); // tandwiel
 u8g2.setFont(u8g2_font_helvB08_tr);
 u8g2.drawStr( 40, 32, "DM-CHECK");
 u8g2.drawStr( 40, 45, "WAIT...");
 u8g2.sendBuffer();          // transfer internal memory to the display  
 delay(1000);

 //Volt check up
 u8g2.clearBuffer();         // clear the internal memory  
 u8g2.setCursor(15, 36);
 u8g2.setFont(u8g2_font_open_iconic_embedded_2x_t);
 u8g2.print(char(73)); // tandwiel
 u8g2.setFont(u8g2_font_helvB08_tr);
 u8g2.drawStr( 40, 32, "VOLTAGE");
 u8g2.drawStr( 40, 45, "CHECK...");
 u8g2.sendBuffer();          // transfer internal memory to the display  
 delay(1000);

//Check compleet
 u8g2.clearBuffer();         // clear the internal memory 
 u8g2.setCursor(45, 45);
 u8g2.setFont(u8g2_font_open_iconic_check_4x_t);
 u8g2.print(char(64)); // vinkje iccon
 u8g2.sendBuffer();
 delay(1000);

```
Yours sincerly,

Jelmer de Jong
```

tablatronix, 2019-12-10:
setConfigPortalTimeout

TomsCircuits, 2019-12-15:
I think your problem is this:

if(!wifiManager.autoConnect()) {
Serial.println("failed to connect and hit timeout");
delay(3000);
//reset and try again, or maybe put it to deep sleep
ESP.reset();
}

When Wifimanager ends with connection failed, the ESP resets. So you are in a loop. Take out the ESP.reset() command and you should be fine.

closed 2020-01-23 by tablatronix