tzapu/WiFiManager #981: OTA web update

tablatronix, 2019-12-12:
I just realized there is no issue for this PR feature

https://github.com/tzapu/WiFiManager/pull/587

I was going to abstract it out into an example, but I might just leave it in and fix it up.

It is available if you add it to the menu

- [ ] add permissions, make hook/route disabled not just menu!
- [ ] find a good way to document this, as it does not work via captive portals! ( maybe even hide it )
- [ ] Better handling of update success

gitcnd, 2020-01-17:
How do I download and install your OTA branch?  Why is it not the default?

tablatronix, 2020-05-10:
- [ ] add check for and get update via http client, users add their own server address, opt url input on ota page, or save in memory

tablatronix, 2020-05-16:
Excessive logging during binary upload of handle root

tablatronix, 2021-01-25:
Merged into master

closed 2023-09-08 by tablatronix