tzapu/WiFiManager #983: Letting code to be executed when waiting connection via portail

matthieuweber, 2019-12-16:
Hello,

I would like to know if there is an option to let's some code to be executed when waiting connection configuration via portal.

I explain:

My Wemos D1R1 (ESP8266) with WifiManager talks with an PIC µcontroller with TTL serial port. 
Sometimes PIC ask Wemos D1R1 its status.

This works only when Wemos D1R1 is in access point mode or if the portal was used to connect to my internet box.

If web portal waits for user to connect to the internet box, I can't execute any code... So I cannot tell the status to my PIC µcontroller.

I tried wifimanager.setConfigPortalBlocking but I get error : setConfigPortalBlocking  not belong to wifimanager class....

I tried with a timeout => Works but I have no opportunity to connect via the portal anymore...

Thanks  !

Matthieu

tablatronix, 2019-12-16:
Use the examples

matthieuweber, 2019-12-16:
Thanks Tablatronix, 

You means this one : AutoConnectWithFeedbackLED.ino 
=>I modify tick() function to respond to external uart commands ?

Thanks,

Matt

matthieuweber, 2019-12-16:
Great ! It's works !

I modified tick() function from AutoConnectWithFeedbackLED.ino to scan UART port and launch a parsing function, so it  responds to PIC asks..

`
void tick()
{
//toggle state
int state = digitalRead(BUILTIN_LED);  // get the current state of GPIO1 pin
digitalWrite(BUILTIN_LED, !state);     // set pin to the opposite state

// Get commands and data from the heater system mainboard via TTL USART (5V RS232 like port):
while(Serial.available() > 0)
  {
  #ifdef  __DEBUG__MSG_VIA_UART_   
    Serial.println("UART chars received!");
  #endif
  Buff_rs232[Buff_rs232_index] = Serial.read(); // Store received chars into Buff_rs232 chart
  if (Buff_rs232[Buff_rs232_index++] == (unsigned char)'\n')  Command_ready = 1;
  }

// Parse commands stored into Buff_rs232[]
Parse_command();
}
`

closed 2019-12-16 by matthieuweber

tablatronix, 2019-12-17:
There is a non blocking example in the development branch