tzapu/WiFiManager #986: Changing Wi-fi Credentials if Wi-fi pre-stored wi-fi not found

rajdpandey, 2019-12-30:
Hi,

As I was searching for the solution to store multiple Wi-Fi credentials, so I can store different locations.

I found this issue #166 

It says we cannot use this library and If we want we can hardcode or create a code like that which can store multiple credentials.

But I think there should be a feature in this library which lets the device ask for new SSID if the old SSID Wi-Fi connection not found. Like there should be an option. 

Wi-Fi credentials should be stored in the EEPROM memory and can be accessed. There should also be an option for removing all credentials from the memory using the web GUI.

If the above one is already implemented please do let me know. I know it will be very difficult to implement and get it to work. But it will be a lot help.

tablatronix, 2019-12-30:
That is literally the whole point of the library and what it already does.

closed 2019-12-30 by tablatronix

rajdpandey, 2019-12-31:
@tablatronix Sorry, I was searching for how to do the above things and I didn't find any examples regarding it. I saw an example that says once we have added credentials through the GUI to change it we have to re-run code with empty creds. And there were some other things. Therefore I posted this here. If by any chance you have examples to do all of the above things Please share with me.

tablatronix, 2019-12-31:
There are plenty of examples included and readme here on github, autoconnect() automatically starts the webportal if connect fails.

rajdpandey, 2019-12-31:
@tablatronix Thanks I will search and try it. I saw the last commit on this repo in examples was 2years ago and also were of max 5 years so I assumed it is not updated and all. It was my mistake. I will start digging and working now. Thanks again.

tablatronix, 2019-12-31:
The still work, also the newest code is in development branch, but it is only alpha