tzapu/WiFiManager #987: Minor issue (compile fails with -Werror)

liebman, 2019-12-31:
The `enable` parameter is unused and if compiling where warnings are treated as errors this causes a failure.

https://github.com/tzapu/WiFiManager/blob/c9665adfaf4ef604e1926ab9a672cce2f485627a/WiFiManager.cpp#L2386-L2390

Ruandv, 2020-01-15:
@tablatronix what is the idea of the enable I see that it was added in Nov but i cant find a reason in the code to try and assist.

JorgeTrincado, 2020-01-22:
Hello! I have shown a little compilation problem (at WiFiManager.h:25:24) with the new version of ESP8266 Arduino core, which redefines HTTP_HEAD as "const char HTTP_HEAD []", maybe a refactor is required, but don't know how to solve it.

This is the complete console output 

```
In file included from C:\Users\Jorge\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,

                 from C:\Users\Jorge\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

                 from C:\Users\Jorge\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266/Arduino.h:32,

                 from sketch\ESP8266WebConfig.ino.cpp:1:

C:\Users\Jorge\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))

                                                                                                                                  ^

C:\Users\Jorge\Dropbox\ArduinoBasic\libraries\WiFiManager/WiFiManager.h:25:24: note: in expansion of macro 'PROGMEM'

 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                        ^

In file included from C:\Users\Jorge\Dropbox\ArduinoBasic\libraries\Arduino-Library-master\src/ThingerWebConfig.h:30:0,

                 from C:\Users\Jorge\Dropbox\ArduinoBasic\libraries\Arduino-Library-master\examples\ESP8266\ESP8266WebConfig\ESP8266WebConfig.ino:2:

C:\Users\Jorge\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WebServer\src/ESP8266WebServer.h:33:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

exit status 1
Error compiling for board Generic ESP8285 Module.
```

Best!

tablatronix, 2020-01-22:
This is already fixed, don't post stuff where it doesn't belong please.

Ruandv, 2020-01-22:
Should the issue not be marked as closed or fixed

tablatronix, 2020-01-22:
I was responding comment, not the issue. anyway its fixed now regardless

commit 87009dc4 (2020-01-22): fixes #987
closed 2020-01-22 by tablatronix