tzapu/WiFiManager #990: WiFiManager fails to reconnect when predefined WiFi credentials are used and default AP is off.

mrdc, 2020-01-08:
### Basic Info

#### Hardware
**WiFimanager Branch/Release:** 0.15.0-Beta
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other
- [x] ESP32-Wroom DevKit

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
- [x] esp32 1.0.4 by Espressif

### Description

Hello,

I'm trying to use WiFiManager with predefined WiFi credentials when I use ESP32 in my office and still be able to connect to WiFi when I'm somewhere else. Code below works when my office AP is ON (predefined credentials in my code):
```
15:27:57.419 -> *WM: [1] AutoConnect 
15:27:57.419 -> *WM: [2] ESP32 event handler enabled 
15:27:57.453 -> *WM: [1] AutoConnect: ESP Already Connected 
15:27:57.453 -> *WM: [1] STA static IP:
15:27:57.453 -> *WM: [2] setSTAConfig static ip not set 
15:27:57.453 -> *WM: [1] AutoConnect: SUCCESS 
15:27:57.453 -> *WM: [1] STA IP Address: 192.168.1.49
15:27:57.453 -> connected...yeey :)
```
WiFiManager is not making it's own AP to configure credentials, so everything is fine. When my office AP is OFF when ESP32 restarts there is a bug: WiFiManager makes it's own AP as it should, I can go to Web interface, etc. But if I do nothing in Web interface, turn on my predefined office AP and restart ESP32 it won't connect to it but instead WiFiManager's AP is created again and again.
To overcome the issues I need to go to WiFiManager's portal -> Info -> Remove WiFi credentials -> then after restart it successfully connects to the predefined office AP.

### Settings in IDE

Module:  		ESP32 Dev Module

### Sketch

```cpp

#include <Arduino.h>
#include "WiFi.h"
#include <WiFiManager.h>

#define WIFI_SSID "MYOFFICE_AP"
#define WIFI_PASSWORD "MYOFFICE_AP_PASSWORD"

void setup() {

WiFiManager wm;
WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
delay(500);
bool res;
res = wm.autoConnect("WiFiManagerAP","WiFiManagerAPPass"); // password protected ap

if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

void loop() {

}
```

**Serial output:**
```
14:34:34.720 -> *WM: [1] AutoConnect 
14:34:34.720 -> *WM: [2] ESP32 event handler enabled 
14:34:34.720 -> *WM: [2] Connecting as wifi client... 
14:34:34.720 -> *WM: [1] STA static IP:
14:34:34.720 -> *WM: [2] setSTAConfig static ip not set 
14:34:34.720 -> *WM: [3] WIFI station disconnect 
14:34:35.047 -> *WM: [1] Connecting to SAVED AP: MYOFFICE_AP
14:34:35.047 -> *WM: [3] Using Password: MYOFFICE_AP_PASSWORD
14:34:35.094 -> *WM: *WM: [2] [EVENT] WIFI_REASON: 202
14:34:35.094 -> [3] WiFi station enable 
14:34:35.094 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
14:34:35.141 -> *WM: [2] Connection result: WL_CONNECT_FAILED
14:34:35.141 -> *WM: [3] lastconxresulttmp: WL_STATION_WRONG_PASSWORD
14:34:35.141 -> *WM: [3] lastconxresult: WL_STATION_WRONG_PASSWORD
14:34:35.141 -> *WM: [1] AutoConnect: FAILED 
14:34:35.141 -> *WM: [2] Starting Config Portal 
14:34:35.141 -> *WM: [2] AccessPoint set password is VALID 
14:34:35.141 -> *WM: [1] XXXXXXX
14:34:35.141 -> *WM: [3] WIFI station disconnect 
14:34:35.141 -> *WM: [3] WiFi station enable 
14:34:35.141 -> *WM: [2] Disabling STA 
14:34:35.141 -> *WM: [2] Enabling AP 
14:34:35.141 -> *WM: [1] StartAP with SSID:  WiFiManagerAP
14:34:35.141 -> *WM: [1] SoftAP Configuration 
14:34:35.141 -> *WM: [1] -------------------- 
14:34:35.141 -> *WM: [1] ssid:             WiFiManagerAP
14:34:35.188 -> *WM: [1] password:         WiFiManagerAPPass
14:34:35.188 -> *WM: [1] ssid_len:         7
14:34:35.188 -> *WM: [1] channel:          1
14:34:35.188 -> *WM: [1] authmode:         3
14:34:35.188 -> *WM: [1] ssid_hidden:     
14:34:35.188 -> *WM: [1] max_connection:   4
14:34:35.188 -> *WM: [1] country:          CN
14:34:35.188 -> *WM: [1] beacon_interval:  100(ms)
14:34:35.188 -> *WM: [1] -------------------- 
14:34:35.656 -> *WM: [1] AP IP address: 192.168.4.1
14:34:35.703 -> *WM: [3] setupConfigPortal 
14:34:35.703 -> *WM: [1] Starting Web Portal 
14:34:35.703 -> *WM: [3] dns server started with ip:  192.168.4.1
14:34:35.703 -> *WM: [2] HTTP server started 
14:34:38.604 -> *WM: [2] WiFi Scan ASYNC completed in 2901 ms
14:34:38.604 -> *WM: [2] WiFi Scan ASYNC found: 9
14:34:38.604 -> *WM: [2] WiFi Scan completed in 2902 ms
14:34:38.604 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
```

tablatronix, 2020-01-08:
Do you expect the esp to automatically reconnect somehow? Do you mean even if you restart it it will not reconnect to the router?

Try erasing your NVS it could be corrupt

mrdc, 2020-01-08:
> Do you expect the esp to automatically reconnect somehow? Do you mean even if you restart the esp, it it will not reconnect to the router?
> 
> Try erasing your NVS it could be corrupt

No, there is no code for reconnect. I was manually pressing the reset button on ESP32. When predefined (SSID&Pass in code) office AP is on, I can press reset 10 times and each time ESP32 successfully connects to it and in logs I see `*WM: [1] AutoConnect: ESP Already Connected` , so everything is fine. But when I turn off office AP, press the reset button, ESP32 reboots, WiFiManager creates AP to configure credentials and if after that I turn on my predefined in code office AP, press the reset button, ESP32 reboots, but fails to connect to my office AP.

tablatronix, 2020-01-09:
interesting. Sounds like corrupt NVS, have you replicated on multiple esps? or tried a full erase?

mrdc, 2020-01-09:
I've tried `#define WM_ERASE_NVS` the same behaviour.  Adding 
```
#include <nvs.h>
#include <nvs_flash.h>
```
and using erase(true) gives me an error. 
At the moment I have only one ESP32 for tests. A new one will be available later this week. Will test my code.

tablatronix, 2020-01-09:
Hmm try a full erase from arduino ide that should do the same

pczekalski, 2020-01-11:
Had similar behaviour. 
Full erase did the trick.

Fasic, 2020-01-22:
i have same problem, but after testing i found that using static (ip/gw/sn) it fails to connect, using DHCP it connects, funny part is it gets same ip from DHCP that i try to use as static...idk why, for now i don't need static ip, but i would like to fix this...and best part is it worked on old project, now it doesn't

tablatronix, 2020-01-22:
Hmm have example code to test?

Fasic, 2020-01-22:
> Hmm have example code to test?

I will try to make "short" version, tomorrow, now it is few files...or if i don't make to make it, i will put on git (but that would be pain to test for you)

tablatronix, 2020-01-23:
Ill try the static ip examples and see, you are using development branch correct?

Fasic, 2020-01-23:
> Ill try the static ip examples and see, you are using development branch correct?

Forget on my comment, i was like debugging for last 2h, and i found out (all works) but in same time, that i am stupid (i didn't plug in mqtt server, that is why i was debugging for 2h)...

For others, 
   -make sure to update library (i was on 0.14, 0.15 all works fine), 
   -plus be sure to check did you wipe all flesh contents (i was 100% sure i did it, but i didn't)

*all fine all good*, sorry for wasting your time *DEV*

tablatronix, 2020-01-23:
heh, no problem, I did the same thing the other day, my ide was `locked` to the wrong build file and uploading the same wrong program over and over, NOT the one I was actually testing..

closed 2020-01-23 by tablatronix