tzapu/WiFiManager #991: Multiple Param Pages

antoniuschan99, 2020-01-09:
Hi,

Is it possible to add multiple Param Pages?

Thank you

antoniuschan99, 2020-01-09:
so far I have something like this to read multiple param values. It's true that there is unlimited EEPROM Read Cycles yes?

void saveUpdateParamsCallback() {
  Serial.println("Saving Update Frequency");
  Serial.println(getParam("updateFrequencyId").toInt());
  
  EEPROM.begin(eepromSize);
  EEPROM.write(address, getParam("updateFrequencyId").toInt());
  EEPROM.commit();
  
  Serial.println("Saving Update Humidity Offset");
  Serial.println(getParam("updateHumidityOffsetId").toInt());

  EEPROM.begin(eepromSize);
  EEPROM.write(address, getParam("updateHumidityOffsetId").toInt());
  EEPROM.commit();
}

tablatronix, 2020-01-09:
I noted this somewhere else, I don’t think I made an issue for it. It is a good idea plus we can let you leave wifi stuff on wifi page,  Right now custom params can only be on wifi or params page.

antoniuschan99, 2020-01-09:
ok great thanks!

antoniuschan99, 2020-01-11:
After saving a parameter, is it possible to retrieve the value that was just saved?

Say the user saved the value, but goes back to the params page. I want the %s value to reflect the value that was just saved. However, it doesn't do that (it always gets the value that was loaded the very first time the device booted)

  const char* setFrequencyValue =
    "<div id='setFrequencyContainer'>"
    "<label for='frequencyId'><b>Update Frequency</b></label>"
    "<input type='radio' name='frequencyId' value='%s' checked>"
    "<input type='radio' name='frequencyId' value='15'>15 Minutes"
</div>";
    
    char write_letter[1024];
    sprintf(write_letter, setFrequencyValue, String(frequencyValue));

  new (&setFrequency) WiFiManagerParameter(write_letter);
    
  wifiManager.addParameter(&setFrequency);

tablatronix, 2020-01-11:
hmm not sure if you can set the param again worth a try just use the new content

antoniuschan99, 2020-01-12:
How would go about doing that? For now I have user save everything at once then will just disconnect.

tablatronix, 2020-01-12:
Hmm might need a callout for that, let me think about it. Not sure we have any hook. You would have to be in non blocking mode

antoniuschan99, 2020-01-13:
ok thanks!