tzapu/WiFiManager #999: Changing AP interface?

bydoctor, 2020-01-28:
I try to change template.h and and html file of template and uploaded the sketch it seems same. 
How we can change interface when we go to 192.168.4.1 page?  I want to configure the Page on my language. But i couldnt

tablatronix, 2020-01-28:
strings are not in that file, they are in the source

bydoctor, 2020-01-28:
So i must change something on github files with fork right?

dknop97, 2020-01-28:
> Then where the file? I just see strings in that extras file. I think i must change something on github files with fork and push the files with my name and change the library.json repository line with my link right? If is not how can i do? Please help bro i just want to edit congifure wifi , configure wifi(noscan) ... page .

You need to change the strings in the _WifiManager.h_ file (see the images in the links below). They are the same strings as the _template.h_ file, in the _extras_ folder. For this, while editing, I use the Sublime Text editor to create my html page and, after editing, I select the part of the HTML code to join it in one line (ctrl + j), following the separation of original tags of model.

Sorry for my English, I'm still improving.
[https://i.imgur.com/mMXMSEm.png](url) -> WifiManager.h
[https://i.imgur.com/dLt6E97.png](url) -> Join Lines command in Sublime Text 3

bydoctor, 2020-01-28:
> You need to change the strings in the _WifiManager.h_ file (see the images in the links below). They are the same strings as the _template.h_ file, in the _extras_ folder. For this, while editing, I [...]

oh man thank you i missed that part now solved my problem thank all of you :)

closed 2020-01-28 by bydoctor

tablatronix, 2020-01-28:
In development branch next version these have been moved to strings.h