tzapu/WiFiManager #1001: AutoConnect and startConfigPortal ?

bydoctor, 2020-01-31:
#### Hardware
**WiFimanager Branch/Release:**
- [x ] Master
- [ ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

How i can use autoconnect and startConfigPortal together. I mean in my project i want my device will be connect wifi with saved credentials and  with button i want open the portal because my customer maybe want to change wifi network with button like our phones . But autoconnect portal just work when connecion lost .  So I just use startConfigPortal  yesterday with pressed button it works properly but when i start today device portal open it suddenly and not close . But my credentials saved before. Why this is happening?

tablatronix, 2020-01-31:
no idea sounds like corrupt credentials, erase flash

bydoctor, 2020-01-31:
> no idea sounds like corrupt credentials, erase flash

Ok thanks i will try.

closed 2020-01-31 by bydoctor