tzapu/WiFiManager #1002: Fatal exception 28(LoadProhibitedCause) Multiple Parameters

rodri16, 2020-01-31:
#### Hardware
**WiFimanager Branch/Release:**
- OTA

**Esp8266/Esp32:**
-  ESP8266

**ESP Core Version:	2.4.0, staging**
-  2.4.0

Hi, I have a problem (OTA Branch on esp8266ex), using multiple parameters. My parameters are 
```
char mqtt_server[40] = "monitor.com"; 
char mqtt_port[6] = "80";
char TimeZone[5] = "-3.0";
char NumDisp[5] = "17";
char mult_V[12] = "448128.6";
char mult_I[12] = "16591.2";
char mult_P[13] = "12136999.39";
```
After full erase flash, everything is ok, but when booting again and finding the file config.jsn it crashes after the 5th parameter. Do you know why?

After full erase
```
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: TimeZone
*WM: [2] Added Parameter: NumDisp
*WM: [2] Added Parameter: mult_V
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: mult_I
*WM: [2] Added Parameter: mult_P
*WM: [2] Starting Config Portal 
```
On reboot:

```
SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af
mounting FS...
reading config file
{"mqtt_server":"monitor.com","mqtt_port":"80","TimeZone":"-3.0","mult_V":"448128.6","mult_I":"16591.2","mult_P":"12136999.39","NumDisp":"16","ip":"192.168.1.231","gateway":"192.168.1.254","subnet":"255.255.255.0"}
parsed json
Fatal exception 28(LoadProhibitedCause):
```
changed to 10 MAX_PARAMS but nothing changed, 

``` #define WIFI_MANAGER_MAX_PARAMS 10```

If I use less parameters everything is OK

tablatronix, 2020-02-01:
Usually running out of memory, we are working on it, see the other issues some hints were to allocate memory for http page variable output to prevent fragmentation

rodri16, 2020-02-04:
Maybe increasing DynamicJsonDocument json(1024);??
Still couldn't make it work

tablatronix, 2020-02-04:
No its memory in wm theres a issue about it

rodri16, 2020-02-04:
So removing PROGMEM could be a solution according to [https://github.com/tzapu/WiFiManager/issues/931](url)

tablatronix, 2020-02-04:
Actually I was thinking of another issue , I forgot about that one. Are you using progmem strings in your params?

rodri16, 2020-02-04:
Actually not in params but in strings_en.h

tablatronix, 2020-02-04:
No those should be fine, we are talking about stack memory

rodri16, 2020-02-10:
[https://github.com/tzapu/WiFiManager/issues/901](url)?? I still don't get what to change to support multiple params

rodri16, 2020-02-17:
I have another weird problem, in config portal after connecting WiFi "Credential saved. Trying.." page doesn't appear, is it because of lack of memory?
Also when calling captive portal  sometimes I have to reload captive portal page many times till I see the WM menu

tablatronix, 2020-02-17:
I was having this issue also, let me se if that is a new bug i introduced

rodri16, 2020-02-18:
Solved after clicking save. Thanks @marcovanoord!
But still cannot always connect directly to configportal... HTTP root not always appearing
```
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  10.0.1.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 1583 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] Portal Timeout In 116 seconds
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] Portal Timeout In 86 seconds
```
So i have to go back and launch again config portal and then portal opens (sometimes) successfully
```
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 10.0.1.1 
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [2] WiFi Scan completed in 1585 ms
*WM: [3] -> 10.0.1.1 
*WM: [3] -> connectivitycheck.gstatic.com 
```

tablatronix, 2020-02-18:
what device?

rodri16, 2020-02-19:
I am using SonOff Pow (r1) Esp8266

tablatronix, 2020-02-19:
What client device is not pulling up captive portal

rodri16, 2020-02-19:
Oh sorry, it's an Android v10

tablatronix, 2020-02-19:
I just merged dev into ota, it should be up to date now