tzapu/WiFiManager #1003: wifiManager.setAPStaticIPConfig

gauix4, 2020-02-06:
Hello everybody
first sorry if my english is not correct :)
I have a question I block on a problem
I would like my IP to be fixed, knowing that I do not know the gateway address, it changes depending on the box where we are connected, it can go from 192.168.1.1 and 192.168.1.254

I use the function wifiManager.setAPStaticIPConfig(int, int , int);
this indicates that I already know my gateway address which is not my case since we connect to several different gateways

my question is the following how to fix the IP without knowing the gateway knowing that it changes according to the box where we are connected

Thank you in advance for your answers ! :)
sorry for my english it is not correct
PS thanks to Google Translate lol

flavio-fernandes, 2020-02-07:
Using [this commit](https://github.com/flavio-fernandes/persistWifiSetting/commit/879c848996c4b1b2cf2163daffe68b2711617838) I can reproduce https://github.com/tzapu/WiFiManager/issues/1003. This is the stack trace from ESP exception decoder:

```
    PC: 0x00000000
    EXCVADDR: 0x00000000

    Decoding stack results
    0x40121376: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
    0x4012e465: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 401
    0x4013346d: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 740
    0x4013870a: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
    0x40121d7f: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
    0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

gauix4, 2020-02-07:
Hello @flavio-fernandes 
thank you very much for your answer but i didn't understand it
I already have to install ESP32 v1.0.4
esp8266 V2.6.3
not against i'm on an esp8266

flavio-fernandes, 2020-02-07:
> Hello @flavio-fernandes
> thank you very much for your answer but i didn't understand it
> I already instaler ESP32 v1.0.4
> esp8266 V2.6.3
> not against i'm on an esp8266
> 

Hello @gauix4 . Sorry for the confusion. My post was not much of a response, but rather
a confirmation that I too am hitting the issue you are describing here. So I thought of posting my findings here instead of opening a duplicate.

I have an ESP32, not an ESP8266 when hitting this. It is very possible that it does not matter which one is used. The [code I'm using](https://github.com/flavio-fernandes/persistWifiSetting) to reproduce this issue uses [Preferences](https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/src/Preferences.h) which I think does not exist in ESP8266. But, that is irrelevant to the issue here.

Hope this makes it a little clearer.

Best,

-- flaviof

> `#include <ESP8266WiFi.h>
> #include <WiFiClient.h>
> #include <DNSServer.h>
> #include <ESP8266WebServer.h>
> #include <WiFiManager.h>
> #include <ESP8266httpUpdate.h>
> 
> const String Firmware_Version={"1.3"};
> 
> void FirmwareUpdate()//OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
> {
> Serial.println("recherche de mise à jour");
> HTTPClient http;
> Serial.println("version du firmwer interne v"+Firmware_Version);
> #define URL_firmware_Version "https://raw.git...... "
> #define URL_firmware_Bin "https://raw.git......"
> 
> http.begin(URL_firmware_Version,"CC AA 48 48 66 46 0E 91 53 2C 9C 7C 23 2A B1 74 4D 29 9D 33");
> delay(200);
> int httpCode = http.GET();
> delay(200);
> String http_version;
> 
> if (httpCode == HTTP_CODE_OK)
> {
> http_version = http.getString();
> Serial.println("version disponible v"+(http_version));
> }
> else
> {
> Serial.println("Erreur verification firmware:");
> }
> http.end();
> 
> if (httpCode == HTTP_CODE_OK)
> {
> 
> ```
> if(http_version<(Firmware_Version))
> {
>  Serial.println("Version du firmware déjà à jour ok");
> }
> 
> if(http_version.equals(Firmware_Version) )
> {
>  Serial.println("Version du firmware indentique");
> }
> 
> if(http_version>(Firmware_Version))
> {
>  Serial.println("Nouvel mise à jour du firmware disponible");
>  WiFiClient client;
[... 161 lines cut ...]
> String req = client.readStringUntil('\r');
> Serial.println(req);
> client.flush();
> if (matchcommand(req,"off")){
> // requête 192.168.4.1/off --> éteins led
> led_off();
> }
> else if (matchcommand(req,"on")){
> // requête 192.168.4.1/on --> allume led
> led_on();
> }
> 
> ```
>   else {
> ```
> 
> Serial.println("invalid request");
> client.stop();
> return;
> }
> client.flush();
> 
> }
> 
> }`

tablatronix, 2020-02-07:
@flavio-fernandes  create a new issue, you are posting an exception, this is not the same as this simple question.

flavio-fernandes, 2020-02-07:
> @flavio-fernandes create a new issue, you are posting an exception, this is not the same as this simple question.

Oh boy, looks like I'm the one who does not know English. :( Very sorry for mixing the wires here. Going to open a new issue and try to clean this mess.

flavio-fernandes, 2020-02-07:
> > @flavio-fernandes create a new issue, you are posting an exception, this is not the same as this simple question.
> 
> Oh boy, looks like I'm the one who does not know English. :( Very sorry for mixing the wires here. Going to open a new issue and try to clean this mess.

Moved it to https://github.com/tzapu/WiFiManager/issues/1006  Sorry folks!

gauix4, 2020-02-08:
no one has an idea ??

tablatronix, 2020-02-08:
I mean what you are asking makes no sense. How can you connect to something you dont know.. do you want to connect via dhcp the first time then change to static afterwards? Then you can jusy use esp function to get gateway.

gauix4, 2020-02-08:
@tablatronix 
if that is useful because i don't know this ip address
my little box is made to go to several houses so the address will not be the same
in my house I know it's 192.168.1.254 but in the other houses I don't know it's not necessarily the same
and my little box communicates with another one so if the IP address is not identical they cannot communicate

tablatronix, 2020-02-08:
I mean if you are connecting to different networks you cannot just use any ip address, it has to be on the same subnet as the gateway. You have to use dhcp

gauix4, 2020-02-08:
@tablatronix 
no actually when i use wifimanager i can't assign a fixed IP address for my server
let me explain
I use wifimanager to create an access point so that the person goes on it and connects to this internet box
until then everything is OK
I also do the same manipulation with the second box
now they are both connected to the same internet box
the problem is that boxing assign an IP address and in my code I can not know in advance so impossible to communicate with the two boxes
so I have to fix their IP address so that it can communicate

IPAddress server(192,168,1,80);
serveur.begin();

i tried putting this in a code it works fine but when i call wifi manager it doesn't work anymore

tablatronix, 2020-02-08:
They should be able to communicate on the same network as dhcp will give them the gateway, thats how it works. 

When you say box you mean 2 esp and 1 router?

gauix4, 2020-02-08:
yes

gauix4, 2020-02-08:
to fix IP address I would need to know the gateway address
except that I don't know her

tablatronix, 2020-02-08:
You could use mdns and host names..

I would after connecting, get the gateway and then set the static ip after connected

`WiFi.gatewayIP().toString()`

gauix4, 2020-02-09:
that's exactly what I want to do but suddenly I have to register WiFi.gatewayIP () in EEPROM and put Wifi.off ();
to put this in wifiManager.setSTAStaticIPConfig (_ip, _gw, _sn);
or can I do it hot?