tzapu/WiFiManager #1004: wifiManager.getWiFiPass() returns an empty string

flavio-fernandes, 2020-02-07:
Using ESP32 version 1.0.4 on development branch.

After successful config, calling wifiManager.getWiFiPass(true) consistently returns
an empty string.

## POST SERIAL OUTPUT !

Code:
```
    preferences.putString(ATTR_WIFI_SSID, wifiManager.getWiFiSSID(true /*persistent*/));
    preferences.putString(ATTR_WIFI_PASS, wifiManager.getWiFiPass(true /*persistent*/));

    Serial.printf("wifiManager id: %s value: %s\n", ATTR_WIFI_SSID, preferences.getString(ATTR_WIFI_SSID).c_str());
    Serial.printf("**BUG** wifiManager id: %s value: %s\n", ATTR_WIFI_PASS, preferences.getString(ATTR_WIFI_PASS).c_str());
    Serial.printf("**BUG** wifiManager id: %s value: %s\n", ATTR_WIFI_PASS, wifiManager.getWiFiPass(true).c_str());
    Serial.printf("**BUG** wifiManager id: %s value: %s\n", ATTR_WIFI_PASS, wifiManager.getWiFiPass(false).c_str());
```
Output:
```
07:07:26.627 -> wifiManager id: wifi_ssid value: exampleSSID
07:07:26.627 -> **BUG** wifiManager id: wifi_pass value: 
07:07:26.627 -> **BUG** wifiManager id: wifi_pass value: 
07:07:26.627 -> **BUG** wifiManager id: wifi_pass value: 
07:07:26.627 -> id: wifi_pass value: SuperSecret valueLen: 40 label: ssid password (to be saved)
07:07:26.627 -> wifiManager parameters saved as preferences
```

if you have a stack dump decode it:
N/A

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather) -- https://learn.adafruit.com/adafruit-huzzah32-esp32-feather
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- ESP32 version 1.0.4

### Description

wifiManager.getWiFiPass(true);  OR wifiManager.getWiFiPass(false);
are not returning the SSID password configured.

### Settings in IDE
N/A

Additional libraries: 
N/A

### Sketch

See:  
https://github.com/flavio-fernandes/persistWifiSetting/commit/5ad667b862638722c2ee032105ed32c535aa1ce7

### Debug Messages
N/A

tablatronix, 2020-02-07:
Are you calling this after wifi has been init? Or is connected?

flavio-fernandes, 2020-02-07:
> Are you calling this after wifi has been init? Or is connected?

Correct. I'm calling it _after_ wifi manager portal was successful in configuring a 
ssid with password.

Not sure if this is relevant here, but I'm _NOT_ using `wifiManager.autoConnect();`
Instead I am using `wifiManager.startConfigPortal();` only.

tablatronix, 2020-02-07:
So you are checking this after saving or something?
Cause wifi does not start automatically on esp32

flavio-fernandes, 2020-02-07:
> So you are checking this after saving or something?
> Cause wifi does not start automatically on esp32

Yes. I invoke wifiManager.startConfigPortal() and procedd with configuring ssid and pass.
All goes well and the function returns as expected. Right after that, I attempt to get the
value of the ssid and the password configured. The ssid returns the right value, but the
password returns empty.

That is what I'm trying to show here: https://github.com/flavio-fernandes/persistWifiSetting/commit/5ad667b862638722c2ee032105ed32c535aa1ce7#diff-05a1eb158f7a5066b9b6d066b4ca593eR109-R112

gauix4, 2020-02-07:
@flavio-fernandes @tablatronix  

do a worry search in wifimanager.ccp ==>  WiFiManager::connectWifi(String ssid, String pass) 
and paste this

if (ssid != "") {
    //trying to fix connection in progress hanging
    ETS_UART_INTR_DISABLE();
    wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();
    res = WiFi.begin(ssid.c_str(), pass.c_str(),0,NULL,true);
    if(res != WL_CONNECTED){
      DEBUG_WM(F("[ERROR] WiFi.begin res:"));
      DEBUG_WM(res);
    }
  } else {
    if (WiFi.SSID() != "") {
      DEBUG_WM(F("Using last saved values, should be faster"));
      //trying to fix connection in progress hanging
      ETS_UART_INTR_DISABLE();
      wifi_station_disconnect();
      ETS_UART_INTR_ENABLE();
      res = WiFi.begin();
    } else {
      DEBUG_WM(F("No saved credentials"));
    }

then in wifimanager.h add

String        getConfigPortalSSID();
    String        getSSID();
    String        getPassword();

here you have the return of ssid and password

tablatronix, 2020-02-07:
No clue it works for me

```php
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
Mode: STA
Channel: 6
SSID (5): Shawn
Passphrase (8): 12345678
BSSID set: 0
YES
SSID: Shawn
PASS: 12345678
```

flavio-fernandes, 2020-02-07:
> No clue it works for me
> 
> ```
> *WM: [2] disconnect configportal 
> *WM: [2] restoring usermode STA
> *WM: [2] wifi status: WL_CONNECTED
> *WM: [2] wifi mode: STA
> *WM: [1] config portal exiting 
> Mode: STA
> Channel: 6
> SSID (5): Shawn
> Passphrase (8): 12345678
> BSSID set: 0
> YES
> SSID: Shawn
> PASS: 12345678
> ```
> 
> Ill probably need more info, like is it actually connecting?

Thanks for trying. Yes, it connects fine and it works as well for me. BUT....
If you call `wifiManager.getWiFiPass(false);` I''m assuming that you would
expect to see `12345678` but you may get an empty string. Can you confirm?

flavio-fernandes, 2020-02-08:
> @flavio-fernandes @tablatronix
> 
> add to wifimanager.ccp
> 
> <snip>
> wifiManager.autoConnect("Hello Word");
> Serial.println(getSSID());
> Serial.println(getPassword());
> 
> here you have the return of ssid and password

Thank you @gauix4 , what you mentioned here worked for me!
I made these 2 changes to try out what you proposed:
- https://github.com/flavio-fernandes/persistWifiSetting/commit/2729af0d35289eadfad60d2f0bb04a790d3b0bc4
- https://github.com/flavio-fernandes/WiFiManager/commit/497364330f065b84fef17383884c1294d2dc6943

And the function returned the password configured! Note I did not have any issues with
the ssid, just the password.

So I don't really know what to do from here, as it seems that the password I expected is just not what it is implemented.

What do you propose? Is this a legit bug? I am okay closing it if that is not important enough.

```
    18:37:56.530 -> *WM: [2] <- HTTP WiFi save
    18:37:56.530 -> *WM: [3] Method: POST
    18:37:56.530 -> *WM: [2] Parameters
    18:37:56.530 -> *WM: [2] --------------------
    18:37:56.530 -> *WM: [2] wifi_pass: SuperSecret
    18:37:56.530 -> *WM: [2] --------------------
    18:37:56.530 -> *WM: [3] Sent wifi save page
    18:37:56.530 -> *WM: [2] process connect
    18:37:58.564 -> *WM: [2] Connecting as wifi client...
    18:37:58.564 -> *WM: [1] STA static IP:
    18:37:58.564 -> *WM: [2] setSTAConfig static ip not set
    18:37:58.564 -> *WM: [3] WIFI station disconnect
    18:37:58.564 -> *WM: [1] CONNECTED:
    18:37:58.564 -> *WM: [1] Connecting to NEW AP: exampleSSID
    18:37:58.564 -> *WM: [3] Using Password: SuperSecret
    18:37:58.564 -> *WM: [3] WiFi station enable
    18:37:58.564 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult...
    18:37:59.070 -> *WM: [2] Connection result: WL_CONNECTED
    18:37:59.070 -> *WM: [3] lastconxresult: WL_CONNECTED
    18:37:59.070 -> *WM: [1] Connect to new AP [SUCCESS]
    18:37:59.070 -> *WM: [1] Got IP Address:
    18:37:59.070 -> *WM: [1] 192.168.10.120
    18:37:59.070 -> *WM: [2] disconnect configportal
    18:38:00.064 -> *WM: [2] restoring usermode NULL
    18:38:00.100 -> *WM: [2] wifi status: Unknown
    18:38:00.100 -> *WM: [2] wifi mode: NULL
    18:38:00.100 -> *WM: [1] config portal exiting
    18:38:00.100 -> wifiManager id: wifi_ssid value: exampleSSID
    18:38:00.100 -> **BUG** wifiManager id: wifi_pass value:
    18:38:00.100 -> **BUG** wifiManager id: wifi_pass value:
    18:38:00.100 -> **BUG** wifiManager id: wifi_pass value:
    18:38:00.100 -> **HACKING** wifiManager getPassword value: SuperSecret
    18:38:00.100 -> id: wifi_pass value: SuperSecret valueLen: 40 label: ssid password (to be saved)
    18:38:00.100 -> wifiManager parameters saved as preferences
    18:38:05.123 -> ets Jun  8 2016 00:22:57
    18:38:05.123 ->
```

tablatronix, 2020-02-08:
I did not test with false.

tablatronix, 2020-02-08:
psk() is literally the same exact code in wm

tablatronix, 2020-02-08:
`    18:38:00.100 -> *WM: [2] wifi mode: NULL` ??
hmm

tablatronix, 2020-02-08:
wifimanager restores the wifi mode that it was at in your sketch, Your mode is WIFI_OFF

if you call autoconnect we change this automatically, you are not calling it. You will need to set a mode in your sketch in setup, as esp32 does not automatically do this like esp8266

flavio-fernandes, 2020-02-08:
> psk() is literally the same exact code in wm

No. Sorry for the confusion. In my case `_pass` is not "" so psk() is not used at all.
What makes it work is the fact the value I was expecting was stored in _pass.
Make sense?

tablatronix, 2020-02-08:
That is just a variable, not the saved password

gauix4, 2020-02-08:
@flavio-fernandes can you drop your code here i take a look and test

gauix4, 2020-02-08:
arduino code:

Serial.print(_ssid);
Serial.print(_pass);

gauix4, 2020-02-08:
add in .ccp

void WiFiManager::handleWifiSave() {
DEBUG_WM(F("WiFi save"));

//SAVE/connect here
_ssid = server->arg("s").c_str();
_pass = server->arg("p").c_str();
DEBUG_WM (_ssid);
DEBUG_WM (_pass);

flavio-fernandes, 2020-02-08:
> @flavio-fernandes can you drop your code here i take a look and test

hello @gauix4 . My code is here:

https://github.com/flavio-fernandes/persistWifiSetting

You can look at branch [issueWifiPass](https://github.com/flavio-fernandes/persistWifiSetting/tree/issueWifiPass) for how I reproduce it.

I am good dealing with this issue now, thanks to your help and @tablatronix . 

Much appreciated.

hayderismael, 2021-02-12:
> @flavio-fernandes @tablatronix
> 
> add to wifimanager.ccp
> 
> ```c++
> String WiFiManager::getSSID() {
>   if (_ssid == "") {
>     DEBUG_WM(F("Reading SSID"));
>     _ssid = WiFi.SSID();
>     DEBUG_WM(F("SSID: "));
>     DEBUG_WM(_ssid);
>   }
>   return _ssid;
>   }
> 
>   String WiFiManager::getPassword() {
>   if (_pass == "") {
>     DEBUG_WM(F("Reading Password"));
>     _pass = WiFi.psk();
>     DEBUG_WM("Password: " + _pass);
>     DEBUG_WM(_pass);
>   }
>   return _pass;
>   }
> 
> 
> then in wifimanager.h add
>     
> String        getSSID();
> String        getPassword();
> 
> arduino code 
> 
> WiFiManager wifiManager;
> wifiManager.autoConnect("Hello Word");
> Serial.println(getSSID());
> Serial.println(getPassword());
> ```
> 
> here you have the return of ssid and password

I have added these lines into .h and .cpp but  I got this error

Error compiling for board NodeMCU 1.0 (ESP-12E Module).
I just need to get the SSID and Password after connecting to the access point, so I can get them from serial print. It will be so so helpful