tzapu/WiFiManager #1006: Exception when using wifiManager.setAPStaticIPConfig()

flavio-fernandes, 2020-02-07:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather) -- https://learn.adafruit.com/adafruit-huzzah32-esp32-feather
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- ESP32 version 1.0.4

### Description

While trying to configure setAPStaticIPConfig using the following code:
```cpp
#define DEFAULT_WIFIMANAGER_PORTAL_IP IPAddress(10,0,1,1)
#define DEFAULT_WIFIMANAGER_PORTAL_GW IPAddress(10,0,1,1)
#define DEFAULT_WIFIMANAGER_PORTAL_MASK IPAddress(255,255,255,0)

wifiManager.setAPStaticIPConfig(DEFAULT_WIFIMANAGER_PORTAL_IP, DEFAULT_WIFIMANAGER_PORTAL_GW, DEFAULT_WIFIMANAGER_PORTAL_MASK);
```
I see the following exception as soon as my laptop connects to the portal:

```
   PC: 0x00000000
    EXCVADDR: 0x00000000

    Decoding stack results
    0x40121376: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
    0x4012e465: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 401
    0x4013346d: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 740
    0x4013870a: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
    0x40121d7f: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
    0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
This is  what I see in the Serial output when it happens:

```
06:35:25.141 -> *WM: [1] SoftAP Configuration
06:35:25.141 -> *WM: [1] --------------------
06:35:25.141 -> *WM: [1] ssid:             ESP32_3312CFA4
06:35:25.141 -> *WM: [1] password:
06:35:25.141 -> *WM: [1] ssid_len:         14
06:35:25.141 -> *WM: [1] channel:          1
06:35:25.141 -> *WM: [1] authmode:
06:35:25.141 -> *WM: [1] ssid_hidden:
06:35:25.141 -> *WM: [1] max_connection:   4
06:35:25.141 -> *WM: [1] country:          CN
06:35:25.141 -> *WM: [1] beacon_interval:  100(ms)
06:35:25.141 -> *WM: [1] --------------------
06:35:25.645 -> *WM: [1] AP IP address: 10.0.1.1
06:35:25.645 -> *WM: [3] setupConfigPortal
06:35:25.645 -> *WM: [1] Starting Web Portal
06:35:25.645 -> *WM: [3] dns server started with ip:  10.0.1.1
06:35:25.645 -> *WM: [2] HTTP server started
06:35:27.953 -> *WM: [2] WiFi Scan completed in 2303 ms
06:35:27.953 -> *WM: [2] Config Portal Running, blocking, waiting for clients...
06:35:40.189 -> dhcps: send_nak>>udp_sendto result 0
06:35:41.442 -> dhcps: send_offer>>udp_sendto result 0
06:35:42.544 -> Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
06:35:42.544 -> Core 0 register dump:
06:35:42.544 -> PC      : 0x00000000  PS      : 0x00060230  A0      : 0x80121379  A1      : 0x3ffb3f70
06:35:42.544 -> A2      : 0x3ffce5cc  A3      : 0x3ffcf088  A4      : 0x3ffce848  A5      : 0x3ffce828
06:35:42.544 -> A6      : 0x0201000a  A7      : 0x0c01000a  A8      : 0x8012121c  A9      : 0x3ffb3f30
06:35:42.544 -> A10     : 0x3ffce5dc  A11     : 0x3ffcf088  A12     : 0x3ffb3f7c  A13     : 0x00000044
06:35:42.544 -> A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014
06:35:42.544 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000
06:35:42.544 ->
06:35:42.544 -> Backtrace: 0x00000000:0x3ffb3f70 0x40121376:0x3ffb3fb0 0x4012e465:0x3ffb3fd0 0x4013346d:0x3ffb4010 0x4013870a:0x3ffb4030 0x40121d7f:0x3ffb4050 0x40088b7d:0x3ffb4080
06:35:42.544 ->
06:35:42.544 -> Rebooting...
06:35:42.580 -> ets Jun  8 2016 00:22:57
06:35:42.580 ->
06:35:42.580 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

This is 100% reproducible. And it never happens if I avoid making the `wifiManager.setAPStaticIPConfig()` call.

### Settings in IDE

Arduino IDE 1.8.10
IDE setting here: https://imgur.com/a/JDwuLlt

WifiManager library, on branch `development`

Additional libraries: 

N/A

Please see https://github.com/flavio-fernandes/persistWifiSetting/commit/879c848996c4b1b2cf2163daffe68b2711617838 for code used to make it happen.

tablatronix, 2020-02-07:
What happen if you use regular const or vars? Not defines

flavio-fernandes, 2020-02-07:
> What happen if you use regular const or vars? Not defines.
> 
> have you tried a plain sketch without wm ?

The same issue happens when I use const, static or plain variables.

See:  https://github.com/flavio-fernandes/persistWifiSetting/commit/17f485ee65d847f014593599490082d9a327143b

```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x401211a2: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x4012e291: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 401
0x40133299: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x40138536: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
0x40121bab: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

```
18:05:19.095 -> here we go
18:05:19.095 -> wifiConfig_init called
18:05:19.204 -> *WM: [1] <form action='/wifi'    method='get'><button>Configure WiFi</button></form><br/>
18:05:19.204 -> <form action='/info'    method='get'><button>Info</button></form><br/>
18:05:19.204 -> <form action='/exit'    method='get'><button>Exit</button></form><br/>
18:05:19.204 ->  
18:05:19.204 -> *WM: [3] allocating params bytes: 20
18:05:19.204 -> *WM: [2] Added Parameter: wifi_pass
18:05:19.204 -> *WM: [2] Added Parameter: mqtt_server
18:05:19.204 -> *WM: [2] Added Parameter: mqtt_port
18:05:19.204 -> *WM: [2] Added Parameter: mqtt_username
18:05:19.204 -> *WM: [2] Added Parameter: mqtt_password
18:05:19.204 -> *WM: [3] Updated _max_params: 10
18:05:19.204 -> *WM: [3] re-allocating params bytes: 40
18:05:19.204 -> *WM: [2] Added Parameter: mqtt_topic
18:05:19.204 -> *WM: [1] Free heap:        263980
18:05:19.204 -> *WM: [1] ESP-IDF version:  v3.2.3-14-gd3e562907
18:05:19.204 -> *WM: [2] Starting Config Portal 
18:05:19.204 -> *WM: [3] WIFI station disconnect 
18:05:19.204 -> *WM: [3] WiFi station enable 
18:05:19.204 -> *WM: [2] Disabling STA 
18:05:19.204 -> *WM: [2] Enabling AP 
18:05:19.204 -> *WM: [1] StartAP with SSID:  ESP32_20A4AE30
18:05:19.204 -> *WM: [1] Custom AP IP/GW/Subnet: 
18:05:19.352 -> *WM: [2] AP has anonymous access! 
18:05:19.352 -> *WM: [1] SoftAP Configuration 
18:05:19.352 -> *WM: [1] -------------------- 
18:05:19.352 -> *WM: [1] ssid:             ESP32_20A4AE30
18:05:19.352 -> *WM: [1] password:         
18:05:19.352 -> *WM: [1] ssid_len:         14
18:05:19.352 -> *WM: [1] channel:          1
18:05:19.352 -> *WM: [1] authmode:        
18:05:19.352 -> *WM: [1] ssid_hidden:     
18:05:19.352 -> *WM: [1] max_connection:   4
18:05:19.352 -> *WM: [1] country:          CN
18:05:19.352 -> *WM: [1] beacon_interval:  100(ms)
18:05:19.352 -> *WM: [1] -------------------- 
18:05:19.844 -> *WM: [1] AP IP address: 10.0.1.2
18:05:19.844 -> *WM: [3] setupConfigPortal 
18:05:19.844 -> *WM: [1] Starting Web Portal 
18:05:19.844 -> *WM: [3] dns server started with ip:  10.0.1.2
18:05:19.844 -> *WM: [2] HTTP server started 
18:05:22.139 -> *WM: [2] WiFi Scan completed in 2304 ms
18:05:22.139 -> *WM: [2] Config Portal Running, blocking, waiting for clients... 
18:05:28.665 -> dhcps: send_offer>>udp_sendto result 0
18:05:29.847 -> Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
18:05:29.847 -> Core 0 register dump:
18:05:29.847 -> PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x801211a5  A1      : 0x3ffb3f70  
18:05:29.847 -> A2      : 0x3ffce2b8  A3      : 0x3ffce4ec  A4      : 0x3ffcd9d4  A5      : 0x3ffcd9b4  
18:05:29.847 -> A6      : 0x0301000a  A7      : 0x0d01000a  A8      : 0x80121048  A9      : 0x3ffb3f30  
18:05:29.847 -> A10     : 0x3ffce2c8  A11     : 0x3ffce4ec  A12     : 0x3ffb3f7c  A13     : 0x00000044  
18:05:29.847 -> A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
18:05:29.847 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
18:05:29.847 -> 
18:05:29.847 -> Backtrace: 0x00000000:0x3ffb3f70 0x401211a2:0x3ffb3fb0 0x4012e291:0x3ffb3fd0 0x40133299:0x3ffb4010 0x40138536:0x3ffb4030 0x40121bab:0x3ffb4050 0x40088b7d:0x3ffb4080
18:05:29.847 -> 
18:05:29.847 -> Rebooting...
```

tablatronix, 2020-02-08:
I am assuming you tried inline also? Ill try to test , in the meantime I am sure there is a example for static ip try that also as this exception seems to make it to the esp lib so see if its a esp bug

gauix4, 2020-02-08:
arduino code:

Serial.print(_ssid);
Serial.print(_pass);

gauix4, 2020-02-08:
ajoute dans .ccp

void WiFiManager::handleWifiSave() {
  DEBUG_WM(F("WiFi save"));

  //SAVE/connect here
  _ssid = server->arg("s").c_str();
  _pass = server->arg("p").c_str();
   DEBUG_WM (_ssid);
   DEBUG_WM (_pass);

tablatronix, 2020-02-08:
Wrong thread

Craigzyc, 2020-05-12:
I have been experiencing this issue. 

It seems like if you do an erase_flash through esptool.py you get some reboots out of it before it starts causing issues. 

Some reboots, it will listen and set the IP as requested (8.8.8.8 in my case) but other reboots it doesn't get properly set, defaults to 192.168.4.1 then reboots on the first client connecting. I see yours does seem to set the IP so this may be similar but unrelated. 
[image]

After a few reboots, I have about a 20% success rate where it will correctly set 8.8.8.8 which fixes the captive portal not appearing on android.

I'm setting the IP like so:
`    wm.setAPStaticIPConfig(IPAddress(8,8,8,8), IPAddress(8,8,8,8), IPAddress(255,255,255,0));
`

tablatronix, 2020-05-12:
hmm, not able to reproduce, ill keep trying

` wm.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));`

[image]

Boguinhos, 2021-03-14:
I can't really check right away but i'm on ESP8266 and i think i'm passing thru the same issue. 
After setting static STA addresses and wm.debug is on, if the ESP does not connect some exception pops up. if Debug is off , ESP runs without crashing but something weird is happening to me. if i always use DCHP connection is perfect. if i am forcing STAIpConfig , the ESP will only connect at second try. first try nothing happens and it seems like its hung up. if i power cycle the device by removing the cable and power it again it will connect with success. just never at first time. if i'm using the computer and serial terminal it seems like this issue does not occur as often, even with wm.debug false.